Ultimaker/Cura #5701: Consider updating Creality Ender-3, CR-10 and CR-10 Mini profiles

probonopd, 2019-05-06:
**Application Version**
Any

**Platform**
Any

**Printer**
Creality Ender-3, CR-10 and CR-10 Mini

**Steps to Reproduce**
Some people say the [Creality Ender-3, CR-10 and CR-10 Mini profiles](https://www.chepclub.com/cura-profiles.html) by Chuck Hellebuyck (from Filament Friday) are superior to the ones that come with Cura. He has given permission to include them in Cura: https://twitter.com/ChuckHellebuyck/status/1125160431541211136

**Additional Information**
I did not run my own comparison tests yet and I would appreciate if someone really experienced in printer profiles could review them and maybe run some tests so that the project can decide, after a discussion, whether including those profiles (or using parts of them) does indeed increase the overall performance of the Creality Ender-3, CR-10 and CR-10 Mini printers.

nallath, 2019-05-06:
We're fine with it either way. If someone makes a pull request for machine profiles, we only check if it's not "sabotage" and if they don't break anything (as that's all we can do without actually having any of those printers)

Some work would need to be done as these would need to be added as "quality" types (and not "quality_changes").

Ghostkeeper, 2019-05-06:
Keep in mind that we've had a minor update to Ender 3 in the meanwhile from someone else as well: https://github.com/Ultimaker/Cura/pull/5456

probonopd, 2019-06-03:
It appears that the most popular Creality profiles right now are in [CreawsomeMod](
https://github.com/trouch/CreawsomeMod/), and it looks like the author is in discussion with the Cura team to have those integrated.

https://github.com/trouch/CreawsomeMod/issues/27

stelgenhof, 2019-06-04:
Perhaps this issue is not the perfect place to discuss, however would like to share a concern I have. 

The community has been active in producing a series of Creality profiles for Cura that all are good in their own ways. And some members (myself included) have submitted PR's to embed some of these settings into Cura. So, this raises some questions, at least for me:

- How do we govern this? 
- There were some initial definitions for a number of Creality printers, then some more improvements, now the Creawsome Mod, and what about tomorrow? 
- The beauty of Open Source is of course that anybody can make/request changes, however how do we know which are the right ones when there are many out there? 
- Who decides which changes are deemed correct and who makes sure these are fine so the end users don't experience a plethora of alterations with each Cura release?

Not expecting a definitive answer here, but just some questions that came to my mind lately :)

(Tagging @nallath and @Ghostkeeper as they may be able to share some experience from their side).

Cheers! Sacha

nallath, 2019-06-04:
The short answer is "I don't know". Our original solution was to simply accept any and all pull requests (wrt to machine definitions), provided that they didn't break anything. Over time this became a bit more strict as we got better at eyeballing the result of certain settings ("Are you very sure you want to put that setting at that value?"), so we got a bit better at providing feedback.

This is the first time that we actually have "competing" profiles (so it's also the first time that an answer is needed). The perspective from Ultimaker is pretty simple; "As long as it doesn't break anything". That's kinda where the original criteria came from. My own opinion is, of course, a bit more nuanced as I want as many people using/contributing/benefiting from Cura (also because that, in turn, has a positive benefit for Ultimaker, albeit indirectly).

What I can say about profile changes is that they are hard to do right. You need a lot of testing and tuning to validate profiles. It's safe to say that we (eg; Ultimaker) aren't going to do that for third-party printers. In part, because we don't have the printers, in part because we have no incentive to do so (It's pretty expensive and we get too little in return). 

There is something to be said for having a single party/person responsible for a given machine/profile, but that does prevent a big advantage of open source development; Multiple people working on something to make it better (because what do we do if two people disagree? Or if the original contributor doesn't spend time on it anymore?)

stelgenhof, 2019-06-04:
@nallath Thanks for the feedback. I totally understand that third party profiles are not of Ultimaker's primary concern. 

Many people in the community don't know that the provided third party profiles are coming from the community itself (and not from Ultimaker). As a result, many point fingers to Ultimaker if something goes wrong with either the print or with Cura itself. Having ungoverned profile changes to Cura may only sustain this perception, which I'd like to see to be avoided :)

nallath, 2019-06-05:
Yeah, same here. Having to explain that certain things are beyond our control (either because we can't or don't want to) is taking up more time than I'd like to spend on it. 

This is also why the beta releases (and people actually coming here to give feedback on it) are so damn important. The biggest problem that we have is that too few people use the betas and feedback is either of poor quality or is only echo-ed in facebook groups (so we don't hear about it until it's too late)

Ghostkeeper, 2019-06-06:
Let me list some desires that we'd like to achieve here:

* Decentralised contributions ("working together").
* Some form of quality control.
* Robustness against sabotage.

This can be done all together. I'd like to propose that we appoint someone in the community that performs a smoke test for certain printers if something changed in that printer. So if someone has a certain printer model at home he can test if the print quality is not completely ruined. Because I love lists, here is another one that summarises the aspects of this printer tester:

# Ultimaker treats these testers the same way as the community translators: The testers get a message when the beta is released that they can perform a smoke test. This prevents overloading them with tests if a lot of changes are made in a short period.
# The tester prints some things with the beta release and sees if the quality is bad. "Bad" meaning that he should test whether the changes made it worse for his sample. This includes all aspects, including slicing time and printing time, not just visual quality. Some subjectivity will be involved.
# If nothing changed in the default g-code output for a certain release, we don't need to inform the tester (other than perhaps to inform that nothing needs to be tested).
# Informing them when the beta is released also gives them a clear deadline: They have 2-3 weeks until the stable version is released.
# If the tester deems the quality to be bad, then the tester should also test with the previous release to see if it wasn't just his printer breaking down. If it was indeed the latest release that made it bad, we have to revert some changes. Sometimes it will be clear immediately what change to revert. Sometimes we may have to revert all changes to the printer profiles since the latest release.
# We should probably post up a standard set of 3 or so models to print for various cases. If the changes are specific to one case (e.g. support settings) then we can leave out the tests for the rest. Be smart about it.
# If the tester has no time to spend on it due to personal circumstances, then we use the current method, trusting the contributor blindly.

stelgenhof, 2019-06-06:
@Ghostkeeper Thank you very much for the quick feedback and the proposal! I think this is perfect.
A little bit of governance is sometimes required so I'd like to volunteer in helping with the tests. :)

Cheers! Sacha

Ghostkeeper, 2019-06-06:
Let me propose that also to the Ultimaker team then. It'll involve a bit of administration on our side as well so there's stakeholders involved there.

Ghostkeeper, 2019-06-20:
We're going through with this plan. @stelgenhof, you wanted to volunteer? Could you e-mail me at r.dulek at ultimaker dot com? Which printer model can you test with?

stelgenhof, 2019-06-20:
@Ghostkeeper Thanks! Yes still like to volunteer if that is possible. As for Creality, I have an Ender-3 that I can use for testing. Also a Prusa MK2.5 if that is needed.

I will send you an e-mail shortly.

Cheers! Sacha