Ultimaker/Cura #5702: 4.0.0 Layer shifting

jonasbjor, 2019-05-06:
**Application Version**
4.0.0

**Platform**
64-bit win10

**Printer**
Tevo Tornado black/gold

**Steps to Reproduce**
Layer shifting occurs at exact same spots every time printing same model. Seen it elsewhere also. The model is Tevo Tornado fang cooler. 

**Actual Results**

**Expected results**

**Additional Information**

[image]

smartavionics, 2019-05-06:
Hello @jonasbjor , please zip up and attach both the gcode and the Cura project (.3mf) file. Thanks.

jonasbjor, 2019-05-06:
[TT_Tornado_Fang_Shroud.zip](https://github.com/Ultimaker/Cura/files/3147639/TT_Tornado_Fang_Shroud.zip)

smartavionics, 2019-05-06:
Thanks for the files @jonasbjor. Have you tried turning off acceleration and jerk control?

Ghostkeeper, 2019-05-06:
Loading that g-code in Cura and setting the colour scheme to Feedrate reveals some bits at similar heights that are printing faster.

Try printing this with the Compensate Wall Overlaps setting disabled.

I think the issue is that the acceleration and jerk is set too high, and then the print head is having too much mass for the steppers.

yet-another-average-joe, 2019-06-10:
I know this is an old post ! But I had the exact same problem when I got my Tornado and wanted to print a very similar fang. It is related to the firmware, not Cura. But Cura can solve most of the problems (with speed, acceleration and jerk control : activate these parameters in Cura).

First of all, this model has to be printed slowly : 40mm/s is enough. Then the firmware EEPROM defaults are stupid. The default acceleration is too high. Set it to 500mm/s². Default jerk is also too high (on mine, it was 20). 8-10mm/s will be fine. If you don't want to mess around with Marlin sources or parameters, enable Cura acc. and jerk.

Another problem : there are lots of overhangs, with small radius and thin walls (elliptic profiles). The material tends to curl, and is sometime hitten by the nozzle.

The shifting occurs along the Y axis. The bed has a huge inertia, and the Y driver Vref is probably too low. I pushed it up to 1.10V insteed of 0.75V, with no overheating issue (use a good IR thermometer, and do it at your own risk !). 

You can also rotate the model 90° (horizontally). The head has less inertia than the bed (by far), evn if the Y motor has better performacnes, it's makes a huge difference.

Ghostkeeper, 2019-06-11:
Would you say that you should set the speed, acceleration and jerk to 40mm/s, 500mm/s² and 10mm/s for every print, or just this sort of print with the sharp V?

yet-another-average-joe, 2019-06-11:
Yes, I use 8mm/s and 500mm/s² for every print, 40 to 60 mm/s for PLA, 40 for PETG, 20 or less for TPU, sharp V or not. These values were flashed (Marlin 2.0), and I activated acc./jerk control in Cura for estimated time.

At first I did'nt understand what happened, and I found answers on a french canadian Youtube channel about jerk tuning for the Tornado (and other similar printers). Jerk always amazed me. In theory, it means an infinite acceleration (and force, therefore torque). I imagine that the steppers inductance softens the speed curve (low pass filter).

I found that with PETG (softer, less or no cooling), jerk and acceleration could be a little more agressive, but I stick with these conservative parameters. It seems now that many Creality CR10 clones users recommand 8 to 10 for jerk, and 500 to 1000 for acceleration.

About curling : I think that with low speeds, the nozzle has time to melt the plastic (ironing), giving better results. 

I shot some pics of my messed up fangs. Probably the same ones that the OP wants to print. A huge collection ! You can see where curling occurs (colored with a red sharpie). All these wrecked prints were printed with Tevo default Marlin parameters. I was begining with 3D printing. And later, I learned about jerk. And I successfully printed that fang.

[image]

WARNING : OFF TOPIC BELOW !!! (I know we are not on a forum)

There's also one design issue with the Tornado and similar printers : the single Z that will not move the gantry evenly on both sides. This makes things even worse (nozzle hitting or dragging the part because of head non horizontal moves). On my side, I nearly solved all the problems with a (self designed) belt driven dual Z. And even with the dual Z, the chinese low cost leadscrews being what they are, on 1 mm, they can be off by 0.1mm (measured using a 50mm/0.01mm DTI ) : this is not neglictible, by far ! These cheap printers are made of bad components, are not square nor parallel anywhere, but with a lot of tuning, and adapted designs, they can give pretty good results... I'm used to machines tools (my lathe is a 2 tons/8 HP monster, and my milling machine an oversized european Bridgeport clone), I had a hard time with this chinese toy ! And had to throw away a lot of plastic !

Ghostkeeper, 2019-06-12:
> I imagine that the steppers inductance softens the speed curve (low pass filter).

Yes, and also the play and flexibility of the gantry. For Ultimaker printers we know that the gantry has a much greater role than the motors in this.

I think we're going to adjust the default jerk to 10 then, down from 20, for Tevo Tornado.

commit b5a59d19 (2019-06-12): Set acceleration to 500 for Tevo Tornado

The users of this printer in #5702 found this to be a better value to prevent layer shifting. Fixes #5702.
closed 2019-06-12 by Ghostkeeper