Ultimaker/Cura #5703: still images from built-in camera

MDA-13, 2019-05-06:
**Application Version**
Cura 4.0.0

**Platform**
HW: Inter Core(TM) i9-7900X CPU @ 3.30GHz
Mem: 32.0 GB (SSD)
C-Drive: WD Blue 3D SATA M.2 500GB
Graphics: NVIDIA GeForce GTX 1060 6GB
OS: MS Windows 10 Home

**Printer**
Ultimaker-003f8a

**Steps to Reproduce**
Switch both PC and UM3 printer on, start Cura, connect Cura to UM3, select "Monitor" tab and click on camera icon on the right side of the upper window.

**Actual Results**
I can see just still images which are refreshing roughly in every 15 second.

**Expected results**
Live view in real time of the working space of the printer.

**Additional Information**
I could reproduce this issue at any time so far.
I think﻿ it can be a sw bug of new Cura version (4.0.0) indeed because I can see the live view of the built-in camera in real time without any problem if I connect to UM3 printer directly via LAN from my browser using its IP-address. Even more, I could see live view in the previous versions of Cura as well.
But if I want to see the working space of the printer via Cura's Monitor page, I will see just still images.

Ghostkeeper, 2019-05-06:
Not able to reproduce this. In my Cura I can see the image refresh at multiple times per second. Not 200FPS fluent but certainly at least 5. It may be an issue with slow wifi (especially since you mention that LAN works fine).

MDA-13, 2019-05-06:
I was not speaking about wireless connection at all.
I cannot see live view through LAN cable if I want to use Cura 4.0.0 monitor camera from my desktop PC.
(The printer is connected to the same LAN of course...)

BUT!!!

Addressed the printer IP directly from a browser (Firefox) everything is okay.
So network has to be perfect.

Ghostkeeper, 2019-05-07:
Ah, that clarifies things.

The stream in Firefox is an actual video stream (compressed with MPEG) while Cura consecutively downloads snapshots. It's expected to be a bit more jittery. However one snapshot every 15 seconds is pretty extreme.

MDA-13, 2019-05-07:
Just for confirmation:

I have measured the frame rate of the camera connected from Cura 4.0.0 monitor using a stopwatch.
From Cura 4.0.0, I can see new pictures in every 12 seconds (0,08Hz - this practically means still view).

At the same time, connecting directly to the printer IP address from Firefox the frame rate can be somewhere around 10Hz (estimated average).

But you need to know that I did not have any problem with this camera using earlier versions of Cura and I did not modify anything in my system since then...
So something has to be changed in the software from this aspect which suggests some kind of software bug I think.

Is is possible to repair it in the next upgrade?

Ghostkeeper, 2019-05-07:
I don't think we can fix it in time for 4.1 since the Connect team is pretty busy and this doesn't happen on our computers yet.

MDA-13, 2019-05-07:
Okay; finally, this is not a showstopper problem...

closed 2024-11-25 by GregValiant