Ultimaker/Cura #5706: [4.0.0] Frequent Corrupt Configuration Errors

si-hb, 2019-05-06:
**Application Version**
4.0.0

**Platform**
Windows 10 on Microsoft Surface Book

**Printer**
UM3

**Steps to Reproduce**
Unsure, problem happens occasionally when booting Cura. I do not recall this happening with earlier versions

**Actual Results**
All my custom material configurations are lost and I have to re-enter them.

**Expected results**
For there to be no configuration errors when launching Cura

**Additional Information**
I add material costs to all my filaments, perhaps this is a clue. 
[Cura Files](https://drive.google.com/file/d/1wSkgR3ULCww0v0LvWZAy5sMiPn5bYa8d/view?usp=sharing)
[Error Message](https://drive.google.com/file/d/1wRjFg_jpat2pnjn4HNCx4DCjjw10pIox/view?usp=sharing)

Liger0, 2019-05-06:
Same. All the custom printers and profiles get corrupted at every restart since 4.0

Ghostkeeper, 2019-05-07:
When starting Cura 4.0 with your settings folder, I get the following message in the log:

```

2019-05-07 11:06:13,003 - ERROR - [MainThread] UM.Settings.ContainerStack.getConfigurationTypeFromSerialized [335]: Could not get configuration type: File contains no section headers.
file: '<string>', line: 1
'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

This looks like the files `definition_changes/Ultimaker+3_settings.inst.cfg` and `Ultimaker+3.global.cfg` got corrupted, meaning that Cura unexpectedly closed while it was writing these files. Maybe it got killed by the task manager, or maybe it crashed completely.

In any case, that data is lost.

Question is really, what caused Cura to crash?

Other question in a different line of search, do you often use multiple instances of Cura at the same time?

si-hb, 2019-05-07:
I do not use multiple instances intentionally, though it's likely I have launched multiple instances accidentally from time to time. The second instance is always closed.

At any rate, I've had the above error at least a half dozen times since using 4.0.0. Never had the problem in earlier versions of Cura.

Ghostkeeper, 2019-05-08:
Well once it's corrupt it's going to stay corrupt until the files are either removed or restored.

Except on first run of a new version, Cura doesn't save anything to file if you don't change anything in the interface, so it shouldn't get corrupt by accidentally opening two instances.

We haven't seen this happen ourselves though. We have no idea yet what causes these files to get corrupt :confused:

Ghostkeeper, 2019-05-09:
It writes configuration files every 10 seconds if there has been a change to any profile or added printer. It puts a message in the log then that it's autosaving. It also writes configuration files when closing Cura (normally, not when crashing) and when upgrading from a previous version during start-up.

si-hb, 2019-05-09:
If I may suggest, perhaps on each config write, rename existing as backup and write new. If Cura detects an onerous config on start it can ask user "Use last known good config?".

Ghostkeeper, 2019-05-10:
We do keep a backup from before upgrading.

It's a bit too big to make a back-up every 10 seconds though.

si-hb, 2019-05-10:
You're writing the backup repeatedly every ten seconds? That would certainly indicate a potential problem to me, and an unnecessary waste of resources and disk write cycles.

Ghostkeeper, 2019-05-13:
No, we're writing only the profiles that have changed. When changing profiles they get marked as "dirty" and then written to disk upon the next autosave, instead of immediately. This saves resources and disk write cycles.

betavr, 2019-06-15:
I just lost my printers' configurations after a power outage. My custom startup/finish g-codes are gone, apparently the cfg files have been overwritten with zeros.  :(

Kira0tori, 2019-12-09:
Hello everyone. Same problem here with Cura 4.4.0. A power outage happened while Cura was open but not used. After a restart, all definitions/profiles were corrupted. I tried to go see in %APPDATA%/Local or Roaming or into Programs/Ultimaker Cura 4.4 but I waw unable to find old defintions or at least to restore them. Is there any way or should I give up already and re-write them from scractch ?
Thanks for your help

Ghostkeeper, 2019-12-10:
Look in %APPDATA%/Roaming/cura/4.4 and see if there is anything there that you need to remove or fix references in. All files there are plaintext.

eman717, 2020-04-20:
This happened to me tonight. I've reentered my Printer's settings a bit ago and it freaked out and corrupted itself again. I've ran CCleaner and updated my graphic driver. This happens for me when hte screen starts flashing a bunch of random colors, and that only happens when i'm running this software. I can't do anything with the PC when taht happens, just a hard reboot. When the PC starts back up and reaches desktop, those config files are all corrupted again.

Ghostkeeper, 2020-04-22:
I'm not really sure what to say. It's not like we are calling a command that makes all the pixels on your screen flash pink and green. Not directly anyway.

If the screen is flashing in random colours, I'd really put the blame somewhere between the graphics driver and the screen. It could be anything from faulty drivers to a broken cable. Cura may be the only application you found that triggers this, maybe because it tends to only re-render parts of the window (something that the Qt framework does for us). I'm not sure if there is something we can really do to help you.

The configuration files are most likely corrupt because your computer was still writing the changes to disk when it freezes or is rebooted.

Thisismydigitalself, 2020-10-18:
I had to come back here and post about corrupt configurations which once in a while happens to me and to so many others. this is going on for years now so i am guessing before 4.x as well. now when i got these errors i want to fix this and not get this message every time i launch Cura. i am running 4.7.0 which i really like. i was searching for the profiles listed in this error message and tried to delete those. i deleted one by one.  first, second, maybe the third one as well were removed from this list which made me believe i can do this, but deleting the next one just made the list twice as long so i restored the files i deleted and the error list got back to how you see. I really hate running Cura and get this annoying error message. please help me get rid of it. thank you.
[image]

closed 2020-10-19 by nallath

fieldOfView, 2020-10-19:
> the startup optimizer plugin that has the option to only delete the broken profiles.

Minor correction: it does not delete anything; it prevents files from being loaded by Cura.

Thisismydigitalself, 2020-10-19:
Must I sign in for downloading and/or add this plugin to Cura?..

Thisismydigitalself, 2020-10-19:
Ok, Found it.
Installed
[image]

Yes,disable please
[image]

Done.
[image]

No more errors.
[image]

Aldo, you're the man.
Thank you.