Ultimaker/Cura #5712: Layer Slider behaves buggy

TheAndi, 2019-05-08:
**Application Version**
3.5 , 3.6 and 4.0

**Platform**
Win 10 64bit 1809

**Printer**
Doesn't matter

**Steps to Reproduce**
Slice a model go into preview of the layers and use the layerslider or arrow keys

**Actual Results**
Slider behaves buggy.
When using the mouse is gets stuck very often and does not reach the end/start.
When using the arrow keys there is always one key stroke of kinda delay. Futhermore I can't reach the top layer at all when pressing the up key. Only when I press down then, it goes up once more (same for the other direction), which is really counterintuitive and annoying.

**Please watch the screen capture, as it is really hard to explain in words:**
https://youtu.be/_LIKsSMHMpA

What I also noticed, is that when I leave the Cura window, the last key stroke gets applied (can be seen right at the end of the video)

**Expected results**
Should work intuitively and without bugs

Ghostkeeper, 2019-05-09:
Unable to reproduce this so far. The layer slider allows me to get to the top layer or the bottom one fine, and there is no delay in the keys pressed. It immediately gets applied for me.

It's a strange issue that you're seeing, especially with that delay. It seems like the last keystroke only gets applied upon pressing the next keystroke. That implies that something in your computer must remember what the last keystroke was, and we have no such variable in our Cura code.

Maybe we're looking at a render update bug? When you rotate the camera, does it update the layer view slider as well?

TheAndi, 2019-05-09:
Yes, it refreshes to the right value whenever I click on something else (e.g. rotating). 
Let me know if you need more info

Another render related problem I have, is when I open minimized Cura (4.0) it shows me a white window (see picture). When I click somewhere, it's back to normal again.
[image]

Ghostkeeper, 2019-05-10:
That sounds like the same rendering update problem to me.

I haven't seen this on any computers here though. I wonder what triggered this to start happening on your computer...