Ultimaker/Cura #5716: [4.0] Cura can't start - File not found error: [WinError 3] The system cannot find the path specified

MafiosoGit, 2019-05-08:
Hi all,
I'm using Windows 10, 1809 version.

I'm not able to start Cura 4.0, 3.6 or 2.4. In the case of the first two mentioned versions, upon starting the executable, i'm greeted with the error as displayed in the image. In the latter version (2.4), simply nothing happens.

[image]

I think it's relevant to say that I have a working drive C: (main) and F:, but no E: drive, although i had it at some point. 
I've tried to troubleshoot the issue myself, but with no luck. I've looked through the registry and environment variables for anything python related pointing to E: and eliminated whatever remained. Uninstalled Cura, reinstalled, but the error persists. 

I downloaded the latest Python version and installed it, and it runs with no issue. I've uninstalled and reinstalled Cura afterwards, but the same error pops up.

More concerning is the fact that I can't find logs anywhere, since I no /cura folder is created inside /AppData/Roaming or /Local or /LocalLow I've looked through the whole two SSD's for cura.cfg, a file i'd expect to be in the /cura folder, but nothing pops up. 

At this point, I don't know what else to do to get Cura working other than completely wiping the SSD's anew. If you think there is a more sensible solution, let me know. 

I still think the issue stems from python and an IDE having been installed in my HDD E: in the past, but then again, there are no more traces of anything python related pointing to it anymore.

Lastly, the same python distribution and IDE was installed in another PC (single ssd) and Cura installed afterwards, and it works fine.

I think this should be a really rare issue, so I thank you for any time you can spare to help me out. I'm at your disposal to try any solution you might think of on my machine.

Thanks for your attention!

ilko-k, 2019-05-08:
W10 v1809 is known to have some system issues. One of them is a problem with rendering special unicode characters. Since I do fansubbing, I couldn't stay with it. At this point, I think you have to change your build. Try the build 1803 instead. But before that, ensure you have all vcredist up to date and dotnetfx 4.7 at least.
Also, cleaning the registry might help with something like ccleaner or wise registry cleaner.

Ghostkeeper, 2019-05-09:
I think something is a bit awry with your Windows user account. Here's how I got at that conclusion.

The error message says that it's going wrong at line 33 of cura_app.py:

https://github.com/Ultimaker/Cura/blob/59b3046c0f04a357593f18434a286832fd072a89/cura_app.py#L33

This creates the `cura` configuration directory. The reason why you couldn't find it is because Cura crashes upon trying to create it. The directory it's creating is the `dirpath` variable which is determined by this line:

https://github.com/Ultimaker/Cura/blob/59b3046c0f04a357593f18434a286832fd072a89/cura_app.py#L25

The `expanduser` function is supposed to turn that `~` into your home directory. So I think that Python thinks that your home directory is on the `E:` drive.

You say you've installed Python yourself as well. You could then test it by running this Python code in a terminal:

```
>>> import os.path
>>> os.path.expanduser("~/AppData/Roaming/cura")
```

What does it say then?

Maybe you'll have to figure out where Windows thinks your home directory is.

MafiosoGit, 2019-05-09:
Hi guys, thanks for the quick replies!

@ilko-k , I've tried all your recommendations, but unfortunately none worked. The only option I have left would be to get windows 1809 on my other machine, in which Cura works fine, to see if it stops working due to the update.

@Ghostkeeper , I've tried the commands you mentioned in python, and in fact, it returns a directory on the E: drive -> E:\\Users\\%username/  
This leads me to believe that, making python point to the correct root directory on the C: drive might solve the problem once and for all, so I'll be investing in that. I've been googling how to do so, but haven't figured out the solution yet. Once I do, and solve the issue, I'll report back.

Thanks again guys! I hope to return with good news.

MafiosoGit, 2019-05-12:
Hi again, 

I've managed to solve the issue after some googling. Here's what I did, in case anyone suffers from the same issue:

0. First of all, clean up your registry using a tool like CCleaner.

1. Confirm the issue is the wrong setting of the HOME variable, by opening a python console and inputting the following commands:

>>> import os.path
>>> os.path.expanduser("~/AppData/Roaming/cura")

In my case, it pointed to an nonexistent drive. If you get a directory that's not a logical one (meaning that Windows can't access it, your problem lies therein. 

2. Open Run and type in lusrmgr.msc.
2.1. Double Click the 'Users' Folder and Double Click your 'User Name'
2.2. Go to the 'Profile' tab and set the Home folder path

3. Create the Environment Variables HOME -> C:\Users\%yourusername% and HOMEPATH -> C:\

4. Open an elevated command prompt and input: set HOME. It should display the values for the HOME, HOMEDRIVE and HOMEPATH variables. Confirm they point to a logical directory.

5. Uninstall and reinstall python, and repeat step 1 to confirm the changes worked.

6. Launch Cura.

This is what I did and I seem to have no more problems. Thanks again to @ilko-k and @Ghostkeeper for helping me out in this!

closed 2019-05-12 by MafiosoGit