Ultimaker/Cura #5722: Quality - Layer Height Entry Window Only Allows 3 Characters to the Right?

Loadmaster1, 2019-05-09:
**Application Version**

4.0
**Platform**

Windows 10
**Printer**

Custom - Tevo Little Monster Delta
**Steps to Reproduce**

Changing "Print Settings" - Quality - Layer Height input 0.1618
**Actual Results**

The value is truncated to 0.161 on this latest version of Cura I can no longer input 0.1618 for layer height value.
**Expected results**

With the earlier versions the data input window would allow a full 4 characters to the right of the decimal place.  With Cura 4.0 I can only input 3 characters to the right of the decimal place.
**Additional Information**

[image]

Loadmaster1, 2019-05-09:
[image]
This is the photo I meant to attach.  It shows the entry fields of issue circled in red.

Ghostkeeper, 2019-05-10:
CuraEngine has always only supported up to 1 micrometre resolution. While previously you could fill in tenths of micrometres there, that wasn't used as it was rounded down to the nearest micrometre. We thought we would limit the front-end to better reflect the value that you're actually going to get.

So this is working as expected.

closed 2019-05-10 by Ghostkeeper

Loadmaster1, 2019-05-10:
Thank you for the reply.  Now I understand.