Ultimaker/Cura #5727: Update profile resets on startup

Inglebard, 2019-05-10:
**Application Version**
Notice in Cura 4.0.0, same issue in 3.6.0 

**Platform**
Arch Linux

**Printer**
Prusa MK2S

**Steps to Reproduce**
- Create custom profile
- Edit "Printing temperature", Printing temperature Initial Layer", "Max comb distance with No Retract".
- "Update profile with current settings"
- Close Cura
- Start Cura

**Actual Results**
- Sometimes all values are resetted. After close/start Cura, without saving the actual profile, value may reappear.

**Expected results**
- Keep change after update profile.

**Additional Information**
- The issue may be related to #5706 #3754
- The value seems not to be reset if there are change but not saved

Dabihu, 2019-05-11:
**Application Version**
Cura 4.0.0

**Platform**
Windows 10 x64

**Steps to Reproduce**
- Create custom profile in an older Cura version
- Make changes in Cura 4.0.0

Repeat:
- Close Cura
- Start Cura

**Actual Results**
Every time random changes are applied to the "original" profile, but not showed that the profile is not the last version.

It seems that cura\4.0\quality\*.cfg doesn't change, only the quality_changes folder change, but these changes randomly applied. Sometimes all, sometimes some or none.

Dabihu, 2019-05-11:
Attached test files
[CuraCustom.zip](https://github.com/Ultimaker/Cura/files/3168911/CuraCustom.zip)

Dabihu, 2019-05-11:
Attached test files
[CuraCustom.zip](https://github.com/Ultimaker/Cura/files/3168924/CuraCustom.zip)

Found a temporary solution. Remove the files from "quality" folder that exists in "quality_changes" folder.

PS: quality/custom_extruder_1_pla_0.2.inst.cfg file missing from last post, so deleted and uploaded again.

Ghostkeeper, 2019-05-13:
I'm unable to reproduce this issue. After restarting, the changes are still in my profile. You say it happens only some times. Maybe you can provide a log file after starting Cura with a case where it didn't restore the settings?

@Dabihu: The settings are expected to import from the older version only once upon starting the new version for the first time. If you go back to 3.6 the changes you make there won't show up in 4.0.

Dabihu, 2019-05-13:
1. run: New settings loaded
2. run: Global: new loaded, extruder 1: old loaded
3. run: Global: new loaded, extruder 1: old loaded
4. run: New settings loaded
5. run: Old settings loaded
6. run: Global: old loaded, extruder 1: new loaded

Attached the full 4.0 folder (within the log)
[4.0.zip](https://github.com/Ultimaker/Cura/files/3173389/4.0.zip)

Inglebard, 2019-05-13:
Here is a video to show the issue.
[videobug.zip](https://github.com/Ultimaker/Cura/files/3174254/videobug.zip)

Inglebard, 2019-05-13:
Here are stderr.log + stdout.log after profile is reset.
[stderr.log](https://github.com/Ultimaker/Cura/files/3174296/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/3174297/stdout.log)

Ghostkeeper, 2019-05-14:
Perhaps this is a hint:
```
2019-05-13 20:32:18,681 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [59]: Loading container prusa_i3_mk2
2019-05-13 20:32:18,766 - WARNING - [MainThread] UM.Settings.SettingDefinition._deserialize_dict [548]: Unrecognised property fabricate_enabled in setting minimum_support_area
2019-05-13 20:32:18,767 - WARNING - [MainThread] UM.Settings.SettingDefinition._deserialize_dict [548]: Unrecognised property intermediate_enabled in setting minimum_support_area
2019-05-13 20:32:18,828 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [270]: The definition cache for definition prusa_i3_mk2 failed to pickle.
```

The two unrecognised properties are probably not the issue. They were in the actual source in the 4.0 release:

https://github.com/Ultimaker/Cura/blob/4.0/resources/definitions/fdmprinter.def.json#L4153-L4154

But I wonder why it is unable to pickle the Prusa (and UMS5) printer...

Dabihu, 2019-06-15:
Version 4.1 bug remains.

@Ghostkeeper: As I wrote before, the problem, that the Cura load both config files:
From "quality" folder and from "quality_changes" folder too.

I think this "definitions" and "definition_changes" too, but I didn't test it. I don't have files in the "definitions" folder.

The problem that, pre 4.0 the "quality_changes" was always empty. Cura used "quality" folder.
After 4.0 the "quality" just read, but changes wrote to "quality_changes".
At load time both loaded, and which tread process later a folder remain in Cura. And this is why the result is unpredictable.

My temporary solution is working (delete same named files from "quality"), but I think is easily fixable in the code for everyone.

Ghostkeeper, 2019-06-17:
If you delete the cache directory in `~/.cache/cura` (while Cura is not running), does that fix the issue?

Inglebard, 2019-06-17:
I tried this some days ago and seems to fix the issue. But I do this when the 4.1 was release so I don't know what fix the issue. Moreover, it's random so wait and see.

Ghostkeeper, 2019-06-17:
When you update to a new version, it should always reset the cache.

Maybe something got corrupt in that cache? Or maybe it got written by root?

Dabihu, 2019-06-18:
@Ghostkeeper: This doesn't help for me.

The content in folder: c:\Users\Dabi\AppData\Local\cura\4.1\cache\definitions\4.1.0 deleted when cura 4.1.0 not running.
After that started Cura 3 times:
1. run: New settings
2. run: New settings
3. run: Old settings

Ghostkeeper, 2019-06-18:
Could you see if there is anything in `%APPDATA%\Local\cura\4.1` or in `%APPDATA%\Roaming\cura\4.1` that you don't have write access to?

Or maybe share the log file of 4.1? The log file you shared previously is a month old.

Dabihu, 2019-06-30:
@Ghostkeeper : Sorry, when I read didn't have time and later forget it.

Log files deleted and started Cura 3 times:
1. run: Old settings
2. run: Old settings
3. run: New settings

Attached the log and config.
[cura.zip](https://github.com/Ultimaker/Cura/files/3343217/cura.zip)

Inglebard, 2020-05-28:
Hi,

On 4.6.1, the issue is still present. Notice 3 times today when adjusting bed, extruder temp and combing.
Any news about this ?

Dabihu, 2020-05-28:
The issue still present in 4.6.1.
Maybe I have a workaround:
0: Close cura.
1: Move all files from "quality" folder to "quality_changes" folder which missing from "quality_changes" folder.
2: Delete the remained files from "quality" folder.
A wrote this almost a year ago, but I don't know, it's good for everyone or not.
For me it's working.

Inglebard, 2020-05-28:
Hi @Dabihu,

Thanks for the workaround. I don't know if I already tried it when you posted it.
So for now with quick tests, it seems to work. I hope it will still work with update and parameters changed.
To be honest due to this bug, I switched to Prusa Slic3r. And i would like to give cura another try. So I reuse cura since 3 weeks now.

If I do not encounter this issue with your solution this next month, I will close the issue. However, it should be great in another version (if the issue is really coming to different files in "quality" and "quality_changes") to properly read/write/transfer/clean files inside "quality" folder and determine what is the role/difference of these two folders. Because two different config files should not load same parameters or at least have a priority order (but can cause this kind of issue if one not load properly, I think).

Ghostkeeper, 2020-06-02:
I'm sorry, we haven't been able to reproduce this bug at all.

Cura should not put anything in the `quality` folder at all. If there is anything in that folder, it means that you modded files manually in there. Without those files, we'll probably not be able to reproduce the bug to see what's happening. The log doesn't mention anything going wrong.

Dabihu, 2020-06-02:
I attached the database and logs almost a year ago. Tried these to reproduce the bug?
Cura didn't use the `quality` folder but was in use before 3.6.0. Files are there from those versions. Not modded by us or any users. And if you make a backup in the cura cloud, the restore wrote back the content. Why not migrate the old files and remove these folders (`quality`,`definitions`)?

Ghostkeeper, 2020-06-03:
This was a change in Cura 3.3, indeed. Before that it used to put the quality-changes profiles in the `quality` folder. However it should migrate those files with the version upgrade automatically. When I upgrade from 3.1 to 3.4 they get properly moved to `quality_changes`. I can't do that with your folder though since it only includes 4.1 profiles, so that's too late to reproduce the bug; the folder is already corrupted.

We should *not* remove those folders though (`definitions`, `quality`, `themes`, `scripts` and `variants`) because users can still use them to mod Cura with hand-written files.

Ultimaker's support team will have told you to upgrade through Cura 3.6 first, due to the nature of how far back we want to keep supporting upgrades. Did you try that? To go back, you'd have to delete or rename the folders with higher version numbers than 3.6.

closed 2020-06-08 by Inglebard