Ultimaker/Cura #5728: Idea to improve Z seam pathing - Suggestion

killint87, 2019-05-11:
I'm having issues with z-seam zits on my Creality CR10 when printing at high speeds and I feel like I've tried everything resolve it. I believe that the problem lies in how Cura (and every other slicer I've used) handles pathing at wall changes.

I have no coding experience but I've attached an image to help illustrate my solution.
I've enabled "Optimize wall printing order" and "coasting" which makes things easier to see.

Fig 1. This is how Cura currently handles inner to outer wall moves. The nozzle moves to the end of the path and makes a hard 90 deg turn to reach the start of the next wall. I've also noticed that Cura treats this tiny move as a travel. In my case, travels are set to 160mm/s. I believe this short,sharp,fast turn is what causes the zits and is exacerbated by higher travel, jerk, and acceleration speeds required to print quickly. It's further exacerbated with "optimize wall printing order" turned on because the zits stack up.

Fig 2. My proposed solution with coasting enabled: Instead of the sharp angle travel at the end of wall 1, run a coasting (NOT travel) path directly to the start of the next wall. This would greatly reduce the sharp angle and resulting vibrations at the start of the outer wall, especially with the currently available option to reduce the coasting speed. Finally, at the end of the outer wall path, follow a shallow angle back to the inner wall to prevent visible bulge at the end of the path before travel or retraction

Fig 3. An example of my method without coasting. Offset the start/stop points of the inner walls based on the desired placement of the outer wall seam (if using z-seam alignment that prefers corners). This still allows for a non-extrusion coasting move that takes a shallow angle path to the start of the next wall. End the outer wall path with a shallow angle coast back into the inner wall before traveling.

This seems to be the only thing preventing my printer from achieving good quality at higher speeds. There isn't always a corner to hide a wall seam. Offsetting the points would also give a very slight strength improvement at corners.

[image]

Ghostkeeper, 2019-05-13:
How would this work when the seams are not stacked up exactly?

[image]

Here I've created a slice of a more complex model where the inner geometry breaks up the outer walls (causing travel moves along the walls), where the seams of the outer walls don't line up and where there are multiple parts of walls.

Ghostkeeper, 2019-05-13:
Coasting is intended to follow the path of the wall for a while because the extra pressure on the nozzle needs to be deposited in the shape of the wall. This can also happen around corners (as can be seen in the above screenshot at the smaller hole on the left, where the coasting path is slightly curved).

In many cases, there is not a wall nearby where you're coasting and in many cases, the two walls will not be printed one after another.

killint87, 2019-05-19:
Those are good points that I didn't think of. From my experience, its mainly the move from the inner to outer wall that has the largest impact so perhaps my initial idea was a bit overly complex.

The coast and the end of the wall should stay the same since it serves a specific purpose.

I think the main thing would be generating an extra travel point for walls that creates a shallow entrance angle for wall start paths prioritizing inner to outer where possible, else performing a standard travel move.

an option for coasting at wall start might also help a little bit with blobs?

Ghostkeeper, 2019-05-20:
Yeah I think the extra travel point to enter at a shallow angle has more promise. Of course you have to be a bit careful to not go outside of the outer border. For instance, if you're printing a cube, you wouldn't want to go in at a shallow angle because that requires you to go outside of the border first. Or maybe you want to go in at a shallow angle relative to the line *before* the line you'll start printing. We can make that.

Ghostkeeper, 2019-05-21:
We decided to defer this implementation for later. We won't have time to implement it any time soon.

LosDelPablos, 2023-06-23:
[image]
maybe this is the solution?

LosDelPablos, 2024-06-22:
Yes please look at my solution. Thank you

GregValiant, 2024-11-24:
I'm cleaning up around here.
The "Scarf Z-seam" function of Cura 5.9.0 would seem to cover this.
Can this be closed?

closed 2024-11-24 by GregValiant