Ultimaker/Cura #5729: Cura completly freeze after start up

Chriss122, 2019-05-11:
**Application Version**
4.00

**Platform**
Windows 10 x64 pro

**Printer**
not printer connected -> selected Ender 3

**Steps to Reproduce**
only start the program.
tried already older version

**Actual Results**
after i open cura, i can work for 10-20seconds. After that, cura froze.
I only can exit it via task manager.
I do not open projects or anything, and the error occours too.
Tryed 4.0.0 and 3.6.0

**Expected results**
no freeze 

**Additional Information**

[image]

i think this wont help, but i cant click on any button.

Specs PC:

RAM: 16GB
CPU: Intel I7-9700k
GPU: RTX2080 Asus
SSD: M.2 Samsung Evo

Tryed to deinstall graphic card driver and re install it. 
also tryed to remove cura und install it again -> admin mode everything.

Ghostkeeper, 2019-05-13:
The log file shows nothing of any relevance, sadly.

Perhaps the problem is that you seem to have tried running Cura as administrator and now some of the files are owned by the administrator account, and Cura is unable to overwrite them?

Chriss122, 2019-05-13:
_> The log file shows nothing of any relevance, sadly._
thanks for your support and time! :)

_> Perhaps the problem is that you seem to have tried running Cura as administrator and now some of the files are owned by the administrator account, and Cura is unable to overwrite them?_

Mhh, that's an option. I started it running as admin after the error occurs every time. Also, I installed 2 versions of cura 3.3.0 and 4.0.0 at the same time. And both are uninstalled and newly installed. I don't find a relation of this issue, cause it appears in every version. It seems like the computer has a problem with cura. But the specs don't speak for that case <.<

Ghostkeeper, 2019-05-13:
Ah well but if access rights were a problem, I guess running it as administrator should prevent the issue (for the run that is done as administrator). But apparently it doesn't.

Chriss122, 2019-05-14:
i completly installed windows again. 
All drivers and windows version are up date.

Still having the same problem as before <.<

i made a movie for the issue.
as you can see, i only start cura and do nothing. After that it crashed (first time i got this issue)
seconds opening of cura, i only load some models and after like 20 seconds, the issue appears.

[https://youtu.be/MtfIRanfKnM](url)

Ghostkeeper, 2019-05-15:
Maybe it's the autosave? That would have the behaviour that it gets triggered after some 20 seconds. It's the last log entry in 3 out of 5 runs in your log file.

It's weird how it still behaves the same even after a complete reinstall of the operating system. That would almost indicate that it's hardware related.

Chriss122, 2019-05-15:
problems with autosave is a good idea !!!
I will try to deactivate this (if it's possible) when I am at home. 
Yesterday after like 100 attempts, it finally worked. But after restarting the computer the issue appears again.

it looks like hardware related...
but in January I build up this computer. So the hardware is not old.
Maybe an INTEL problem... 
also interesting is that the logs are not showing a problem with the software. 

When I am at home, I will try the autosave option and answer to the topic later.

Chriss122, 2019-05-15:
is there an option for deactivating auto-saving? 
or can I only change by changing an edited file?

jorisvergeer, 2019-05-15:
I also have this issue. To me it started happenign after a windows update last week. 
I see that there are new updates available (including graphics related ones). I really hope that this fixes my current issues.

Liger0, 2019-05-15:
It has happened to me on 4.1 beta.
I let it run and after some minutes it unfreezed.

Ghostkeeper, 2019-05-16:
> is there an option for deactivating auto-saving?

There is no option in the GUI for this, but there is a preference. If you go to `C:\Users\<Chriss122>\AppData\Roaming\cura\<version>` and open `cura.cfg` with a text editor, you can add the preference `autosave_delay = 999999999` under the `[cura]` category. This will cause it to autosave every 999999999ms (every 2 weeks or so, but the timer restarts every time you launch Cura).

However, the autosave also gets triggered when you close Cura. If it crashes then, Cura will be unable to save any of its configuration.

Chriss122, 2019-05-17:
mhh, when I am editing autosave_delay = 999999 cura doesn't start anymore. 

`2019-05-17 18:03:28,083 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]: An uncaught error has occurred!
2019-05-17 18:03:28,085 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: Traceback (most recent call last):
2019-05-17 18:03:28,086 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
2019-05-17 18:03:28,087 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
2019-05-17 18:03:28,088 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
2019-05-17 18:03:28,089 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
2019-05-17 18:03:28,090 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "__startup__.py", line 12, in <module>
2019-05-17 18:03:28,092 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
2019-05-17 18:03:28,093 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
2019-05-17 18:03:28,094 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
2019-05-17 18:03:28,095 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
2019-05-17 18:03:28,096 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "Console.py", line 24, in <module>
2019-05-17 18:03:28,097 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "X:\4.0\build\inst\bin\cura_app.py", line 132, in <module>
2019-05-17 18:03:28,098 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 758, in run
2019-05-17 18:03:28,100 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "X:\4.0\build\inst\lib\python3.5\site-packages\cura\AutoSave.py", line 19, in __init__
2019-05-17 18:03:28,101 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: TypeError: setInterval(self, int): argument 1 has unexpected type 'str'`

Chriss122, 2019-05-18:
at the moment cura works okay, quit fewer freezes. 
But anyway I want to add those lines into my code. Tried a bit, but still doesn't work :(

Ghostkeeper, 2019-05-20:
Oops, that crash has been in there for 3 years, apparently. Nobody ever edits those settings really. Fixed that for 4.2.

kenjanzen, 2019-12-05:
you said this fixes 4.2. what about 4.4, is the same true for it. can we get to it the same way as deactivating auto-saving?

Ghostkeeper, 2019-12-05:
Since Cura 4.4 is based off 4.2, the fix should also be in Cura 4.4.

I think you're seeing a different error, Kenjanzen. The white screen could be caused by a multitude of things.

kenjanzen, 2019-12-10:
I can't make these changes as the only cura.cfg file is empty. I
cannot find AutoSave.py or cura/AutoSave.py as in your github. Where would
I find it or what do I need to see & change it (software, Etc.)
Any help would be appreciated.
Thank You
Ken Janzen

Ghostkeeper, 2019-12-10:
> I cannot find AutoSave.py or cura/AutoSave.py as in your github. Where would I find it or what do I need to see & change it (software, Etc.)

You won't find that file in your installation if you're using Windows. There is no way to modify it for you. But I modified it in Cura 4.2 (and 4.3 and 4.4 and every future release) so you won't need to.

The `cura.cfg` file being empty is an indicator of more serious problems going on. Maybe your firewall is blocking Cura?

kenjanzen, 2019-12-28:
I still can't get past the white screen where cura should be,

kenjanzen, 2020-02-03:
Any new Ideas as I still cant run cura 3.*.* or 4.*.*. Please help me some
more or tell me who to talk to.

mahtDFR, 2020-02-03:
> Any new Ideas as I still cant run cura 3.*.* or 4.*.*. Please help me some more or tell me who to talk to.
> […](#)

@kenjanzen The discussion here fixed something similar to the white screen you are encountering.
#7019 

Maybe of use to you? Otherwise more info is needed.

kenjanzen, 2020-02-07:
Tried this with no change. Still cant use it.

kenjanzen, 2020-02-14:
Is there any files I can send you that might help me run Cura 3 or newer on
Windows 10?
I can run Cura 15.04.6 but don't have the new advances to use.
I really like Cura 4.* so please help me.

mahtDFR, 2020-02-14:
@kenjanzen This issue isn't reproducible for us with the information provided, so there's not much we can do to help. The issue you're describing is also off topic for this ticket, so please open another for your specific issue.

closed 2020-06-19 by no-response[bot]