Ultimaker/Cura #5730: Cura 4.0.0 Major performance issue in PREVIEW mode. Slow rotation, mouse pointer etc.

pcotegithub, 2019-05-11:
Just installed Cura 4.0.0 on a fast computer. I am trying to rotate a model on PREVIEW and I am experiencing extreme performance degradation. I never had problem before using a modest performance computer using Cura 3.6.

Now on a fast computer, the same model is extremely difficult to manipulate, rotate in PREVIEW mode.

Tried with other models and the same result was observed.

fieldOfView, 2019-05-13:
> on a fast computer.

Please elaborate: what CPU, what GPU, what OS?
Have you tried forcing the layer view to Compatibility Mode in the General pane of the preferences?

diegopradogesto, 2019-06-20:
Hi @pcotegithub , this ticket is open and waiting for more info from your side for more than one month. Are you still experiencing this issue? Could you try in Cura 4.1 to see if it's better?

I'll close this ticket because of inactivity, but please feel free to reopen it if you still have the problem and we need to investigate a bit more.

closed 2019-06-20 by diegopradogesto