Ultimaker/Cura #5731: [4.0.0] Resetting settings on broken machine settings doesn't work.

thopiekar, 2019-05-12:
**Application Version**
4.0.0.

**Platform**
Doesn't matter (I think)

**Printer**
Doesn't matter

**Steps to Reproduce**
Don't know. Was not editing any settings but developing on plugins.

**Actual Results**
Profiles fail to load. That triggers the printer setup window and a crash follows.
Resetting settings are offered in the back, but can not reach there, because the printer setup is in focus.

**Expected results**
You can reset settings using Cura.

**Additional Information**
[image]

[image]

Ghostkeeper, 2019-05-15:
I don't know what caused the broken configuration. Somehow the profile has no ID, but that's expected to be added to the metadata of every profile, always upon deserialising and constructing.

We've discussed the issue with not being able to reset the settings and we won't give it a priority for the next release because it's a very specific case to have it crash during the welcome screen. We've given it a short try but the fix is not easy since the modal window is a Qt window which always goes on top regardless of Z ordering.

thopiekar, 2019-05-15:
No problem. At least you are aware that this situation can happen! 😃

closed 2023-06-24 by github-actions[bot]