Ultimaker/Cura #5732: Mac Update than Cura 4.0 too slow...

andyrok, 2019-05-12:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

andyrok, 2019-05-12:
Cura 4.0 stable
Apple MacOs Mojave 10.14.5

Cura after i update my mac are too slow ....every thing

fieldOfView, 2019-05-13:
Likely a duplicate of #4505

Should be fixed in the upcoming Cura 4.1. A patched version of Cura 4.0 can be found here:
https://github.com/Ultimaker/Cura/issues/4505#issuecomment-482270203. Could you test if that indeed fixes your issue?

andyrok, 2019-05-13:
> #4505 (comment) <https://github.com/Ultimaker/Cura/issues/4505#issuecomment-482270203>. Could you test if that indeed fixes your issue?

thanks!!!!!
your patch is OK!!!!

Andrea Rossi

Le informazioni contenute in questo messaggio e negli eventuali allegati sono da considerarsi riservati ed il loro utilizzo è consentito unicamente al destinatario ed unicamente per le finalità sopra indicate. Qualora riceveste il presente messaggio per errore e non ne siate i destinatari, Vi preghiamo di darcene notizia via e-mail, di astenervi dal consultare eventuali files allegati, di cancellare il messaggio dal Vs. sistema informatico. Costituisce comportamento contrario ai principi dettati dal D.Lg. 196/2003 il trattenere il messaggio, diffonderne il contenuto, inviarlo ad altri soggetti, eseguirne copia in tutto od in parte, l’utilizzo da parte di soggetti diversi dal destinatario e per finalità diverse da quelle sopra indicate.
—
Notice: This message and any attachments may be of a confidential nature or may require protection for other reasons. Should you not be the intended recipient of this message or should you have received this message by mistake, you are not allowed to forward, copy or disseminate the content of the message in any form. Should you have received this message by mistake, please inform the sender and delete the message along with the enclosures.

closed 2019-05-14 by Ghostkeeper