Ultimaker/Cura #5733: Cura doesn't perform Custom Start G-code while printing *.gcode files

sorsdt, 2019-05-12:
**Application Version**

4.0.0
**Platform**

Windows 7 64bit
**Printer**

**Steps to Reproduce**

Open *.gcode for printing
**Actual Results**

Custom Start G-code in machine settings for the selected printer doesn't execute
**Expected results**

Execution of custom G-code
**Additional Information**

sorsdt, 2019-05-12:
I'm using bl-touch calibration sensor so I've added G29 command in custom G-code. When I sent to print *.gcode file I've realized that calibration was not performed. So it would be great to add the ability to run some commands anyway or in special cases like printing G-code files or\and after slicing

fieldOfView, 2019-05-13:
When you open a gcode file, that file is sent to the printer as is; no modifications to the print settings can be made, the start gcode is not changed, and no postprocessing scripts are executed.

closed 2019-05-13 by sorsdt