Ultimaker/Cura #5734: [3.6.0] It is not sliced if the initial layer is high.

yoch2015, 2019-05-13:
Hello.
In order to print nylon, since a bed of 1 mm is covered with a phenol board, if it is sliced ​​by plus 1 mm to initial layer height, printing will start from the middle without being sliced ​​about 1 mm.
Is there a solution?
The photo is a layer view with the first layer selected, but the part that is not printed below is already displayed.
[image]

fieldOfView, 2019-05-13:
Try the Z Offset plugin.

yoch2015, 2019-05-13:
Thank you.
I installed the plug-in and set it to 1.1mm.
The generated g code was started at 1.25 mm, so it was resolved.
[image]

closed 2019-05-13 by yoch2015

Ghostkeeper, 2019-05-13:
Alternatively, disable the setting "Remove empty first layers".