Ultimaker/Cura #5737: Temperature

shape667, 2019-05-13:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2019-05-14:
Yeah, it's getting warmer isn't it? I'm looking forward to summer as well.

I don't see that as an issue though.

Ghostkeeper, 2019-05-14:
Oh, sorry. I see that you filled in the form in the comment `` tags, which is why it's not showing up here. I'll edit the post to make it visible.

Ghostkeeper, 2019-05-14:
It's saying `M104 S220` because that's the value for the "Printing Temperature Initial Layer" setting. The authors of your printer's definition wanted the initial layer printing temperature to be 5 degrees higher than the rest.

You'll find that later in the g-code it sets the temperature to 215 and uses that for the majority of the print.

closed 2019-05-14 by Ghostkeeper