Ultimaker/Cura #5739: Cura 4.0 custom settings menu is not rendering correctly on Arch, even in AppImage

MatthewCroughan, 2019-05-14:
I can't find an issue relating to this, and I've gotten fed up and decided to at least get a conversation started. 

This is the view from a fresh install on Arch from the official repositories. I've not yet tested this on another Arch system.

[image]

I swear I see some qml errors in the log at some stage, I have to keep monitoring it. It's a shame I can't pipe the log into into a text file with `cura > output.txt` for some reason, it ignores stdout I think? I've never encountered that before, so if someone could also help me understand how to record a log that would also be greatly appreciated. Thanks!

https://gist.github.com/MatthewCroughan/d7c57c31587c569efd09db5fdb17e8d3

MatthewCroughan, 2019-05-14:
Another problem I've encountered after following advice online and removing the contents of `.local/share/cura` and `.config/cura` is that this now shows on startup, and there's no way to click "Yes" to it as it suggests.

[image]

Ghostkeeper, 2019-05-15:
Cura from the Arch repositories causes it to run on a more modern version of Qt than what Cura was designed for. Cura was made for Qt [5.10.1](https://github.com/Ultimaker/cura-build-environment/blob/6a1ddf17c29135da43e607609414e71a3cce2883/projects/qt.cmake#L2). Unfortunately, upgrading to a newer version of Qt in our builds would drop support for some old-but-still-fairly-common MacOS versions. Also unfortunately, Qt doesn't seem to be fully backwards compatible or we are unknowingly relying on a bug to render some things correctly.

> I swear I see some qml errors in the log at some stage, I have to keep monitoring it.

Some QML warnings are expected. We were once in a state without any warnings and it was glorious since every warning indicated a bug in our software. However we've let our guard down once, a warning slipped in that was acceptable, and since then we are not alerted any more of the bright orange colour in our log since the warnings are always there :disappointed: We have some binding loops in QML at the moment. Perhaps those are the ones that render differently?

> It's a shame I can't pipe the log into into a text file with `cura > output.txt` for some reason, it ignores stdout I think? I've never encountered that before, so if someone could also help me understand how to record a log that would also be greatly appreciated. Thanks!

As for piping the QML warnings, try piping stderr. Qt outputs these warnings (out of our control) so we can't put them in the normal Cura log.

> Another problem I've encountered after following advice online and removing the contents of `.local/share/cura` and `.config/cura` is that this now shows on startup, and there's no way to click "Yes" to it as it suggests.

Did you remove those files while Cura was running? It's going to write some of the files back upon closing then and some not, resulting in a broken configuration. Removing `~/.local/share/cura`, `~/.config/cura` and `~/.cache/cura` while Cura is closed should result in Cura running completely from factory defaults. (I've yet to see the cache cause a problem though.)

MatthewCroughan, 2019-05-15:
@Ghostkeeper Thanks so much for getting back to me so quickly.

`qml: TableViewSelection: index out of range` is spat out along with font errors when loading the appimage, logged here

https://gist.github.com/MatthewCroughan/9400b99622234ce084844315c49db4f6

it renders like this 
[image]

the only menu that renders seemingly correctly is recommended, custom is unusable.

[image]

MatthewCroughan, 2019-05-15:
I just had a friend test it on his Arch machine, and we're running the same version of qt5-base. It works fine.

Ghostkeeper, 2019-05-17:
I'm running with Qt 5.12 on Ubuntu 19.04, and for me that screen also renders fine. Some others, like the slice estimates tooltip, have a similar problem for me.

Since the bug only occurs in the custom settings menu and it looks like it thinks the height is 0, it could be related to the manual resizing of the window. What happens when you drag the 3 horizontal dots downward?

MatthewCroughan, 2019-05-17:
@Ghostkeeper The appimage worked fine for me, for a moment, on a first run. However on the second run it produces the same glitching, albeit with a centered font on the top right sign-in and marketplace buttons.

MatthewCroughan, 2019-05-17:
Dragging the 3 horizontal dots downwards seems to reveal some of the menu, yes, interesting.

MatthewCroughan, 2019-05-17:
[image]
However some of the menu items are unpopulated :(

Ghostkeeper, 2019-05-17:
> Do you know if there is a way to run applications with different versions of qt5? Is that seriously not possible?

Cura's code will import the `PyQt5` Python library and use whatever it gets. However in the AppImage it's provided with our own build of Python and PyQt5, so it should only get the one in that AppImage file.
Qt is generally backwards compatible so upgrading should be safe in theory. In practice though we notice that it's not always compatible on some details and also sometimes suddenly drops support for older operating systems. So in general we only tend to upgrade if losing OS support is acceptable and we're in need for a bugfix or feature in later versions. That's basically what you seem to be running into now, it not being fully backwards compatible.

> @Ghostkeeper The appimage worked fine for me, for a moment, on a first run.

The height of that window gets stored in the user preferences (because the user can resize it). Maybe something got corrupt there.

phrxmd, 2019-05-19:
For what it's worth, I have a similar issue on Ubuntu 19.04 with Qt 5.12.2 and a HiDPI display (Thinkpad X1 Yoga). With the PPA by @thopiekar I run into scaling issues. Some UI elements are inaccessible, others have their labels misplaced (the "Marketplace" and "Sign In" buttons), others are visible but don't respond to clicks (such as the custom, or the "Allow" button in the "Collecting Data" pane. The visual behaviour is a bit different for different values of ```QT_SCALE_FACTOR```. 

With the AppImage it seems to work, apart from two ```qml: TableViewSelection: index out of range``` errors that keep getting spat out during startup.

thopiekar, 2019-05-19:
Please retry with the master PPA (if you also know how to roll back to stable PPA of course). Got it locked on the 4.1 plus an hack on the environment variables.

That should prevent some errors which come from the systems theming for example.

Depending on the time I have, I'll roll out the 4.1 beta to stable in the following days. But will ask the Ultimaker staff whether there is something major that needs to be fixed in Cura first.

phrxmd, 2019-05-20:
> @thopiekar Please retry with the master PPA (if you also know how to roll back to stable PPA of course). Got it locked on the 4.1 plus an hack on the environment variables.

The version from the master PPA fails to start and throws an exception:
```
$ sudo apt install cura
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  cura-plugin-curadrive cura-plugin-gcodereader cura-plugin-postprocessing
  cura-plugin-sliceinfo cura-plugin-ultimakermachineactions cura-plugin-um3networkprinting
  cura-plugin-usbprinting cura-plugin-x3dreader cura-plugins-all
Die folgenden NEUEN Pakete werden installiert:
  cura cura-plugin-curadrive cura-plugin-gcodereader cura-plugin-postprocessing
  cura-plugin-sliceinfo cura-plugin-ultimakermachineactions cura-plugin-um3networkprinting
  cura-plugin-usbprinting cura-plugin-x3dreader cura-plugins-all
0 aktualisiert, 10 neu installiert, 0 zu entfernen und 10 nicht aktualisiert.
Es müssen 15,7 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 74,3 MB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] 
Holen:1 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [12,2 MB]
Holen:2 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-curadrive all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [29,4 kB]
Holen:3 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-gcodereader all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [9.192 B]
Holen:4 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-postprocessing all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [29,1 kB]
Holen:5 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-sliceinfo all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [8.340 B]
Holen:6 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-ultimakermachineactions all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [6.004 B]
Holen:7 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-um3networkprinting all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [3.382 kB]
Holen:8 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-usbprinting all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [14,1 kB]
Holen:9 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugin-x3dreader all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [12,2 kB]
Holen:10 http://ppa.launchpad.net/thopiekar/cura-master/ubuntu disco/main amd64 cura-plugins-all all 1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1 [2.608 B]
Es wurden 15,7 MB in 1 min 14 s geholt (211 kB/s).                                         
Vormals nicht ausgewähltes Paket cura wird gewählt.
(Lese Datenbank ... 327612 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../0-cura_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-curadrive wird gewählt.
Vorbereitung zum Entpacken von .../1-cura-plugin-curadrive_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura-plugin-curadrive (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-gcodereader wird gewählt.
Vorbereitung zum Entpacken von .../2-cura-plugin-gcodereader_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura-plugin-gcodereader (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-postprocessing wird gewählt.
Vorbereitung zum Entpacken von .../3-cura-plugin-postprocessing_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura-plugin-postprocessing (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-sliceinfo wird gewählt.
Vorbereitung zum Entpacken von .../4-cura-plugin-sliceinfo_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura-plugin-sliceinfo (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-ultimakermachineactions wird gewählt.
Vorbereitung zum Entpacken von .../5-cura-plugin-ultimakermachineactions_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura-plugin-ultimakermachineactions (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-um3networkprinting wird gewählt.
Vorbereitung zum Entpacken von .../6-cura-plugin-um3networkprinting_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura-plugin-um3networkprinting (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-usbprinting wird gewählt.
Vorbereitung zum Entpacken von .../7-cura-plugin-usbprinting_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
Entpacken von cura-plugin-usbprinting (1:4.1~201905200949~rev4001~pkg310~ubuntu19.04.1) ...
Vormals nicht ausgewähltes Paket cura-plugin-x3dreader wird gewählt.
Vorbereitung zum Entpacken von .../8-cura-plugin-x3dreader_1%3a4.1~201905200949~rev4001~pkg310~ubuntu19.04.1_all.deb ...
[... cut ...]
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 80, in <module>
    from cura.Machines.Models.DiscoveredPrintersModel import DiscoveredPrintersModel
  File "/usr/lib/python3/dist-packages/cura/Machines/Models/DiscoveredPrintersModel.py", line 11, in <module>
    from UM.OutputDevice.OutputDeviceManager import ManualDeviceAdditionAttempt
ImportError: cannot import name 'ManualDeviceAdditionAttempt' from 'UM.OutputDevice.OutputDeviceManager' (/usr/lib/python3/dist-packages/UM/OutputDevice/OutputDeviceManager.py)
```

thopiekar, 2019-05-20:
Strange. Will take a look into it later. It is built from the 4.1 branches, so it could be that some package is not up-to-date. So waiting for a day for the next package should fix it.
But the Ultimaker team can probably tell more about the reasons for this error message. 🤓

phrxmd, 2019-05-20:
> @thopiekar Strange. Will take a look into it later. It is built from the 4.1 branches, so it could be that some package is not up-to-date. So waiting for a day for the next package should fix it.
> But the Ultimaker team can probably tell more about the reasons for this error message. nerd_face

I got it to work, It was a dependency issue. I had installed Cura from the stable PPA with its dependencies, then removed it, but hadn't autoremoved all of its packages. When I installed the 4.1 version from the master PPA, it was happy with the dependencies from the stable PPA, but then threw an error on startup. 

The UI on the 4.1 version works for me. Thanks for the good work, can't wait to see it in the stable version.

thopiekar, 2019-05-20:
Glad to hear this. Good motivation to move the beta to stable.

Ghostkeeper, 2019-05-20:
> But the Ultimaker team can probably tell more about the reasons for this error message.

The `ManualDeviceAdditionAttempt` class was apparently added to Uranium in Cura 4.1. It's currently on the 4.1 and master branches. So it seems to me like the Uranium version is still on 4.0 or lower.

Ghostkeeper, 2020-03-19:
I believe this is fixed now, since we had some PRs fixing issues with more modern Qt versions?

lonetech, 2020-05-22:
I just did a [workaround](https://github.com/Ultimaker/Cura/pull/7804) for it (ActionButtons in general were broken, including the welcome screen). However, my patch isn't complete; the Custom/Recommended settings toggle is now usable, but definitely badly sized and placed. I'm using Qt 5.12.