Ultimaker/Cura #5741: [3.6] Post processing plugin - Tweak at Z changing gradually instead of by steps

SecAficionado, 2019-05-15:
**Application Version**
3.6

**Platform**
Ubuntu Linux 18.04 LTS

**Printer**
Custom printer for Anet A8

**Steps to Reproduce**
Used Temp Tower test and tried to modify the temperature for each section. Added ten instances of the tweak Z script, selected trigger as layer number starting at 8, for 50 layers, then 58, for another 50 layers and so on for the total height of the model. The temperature of the extruder was set in decrements of 5, from 225 to 185.

**Actual Results**
The printer started changing the temperature at layer 8, increasing gradually each layer to reach 225 at layer 58 and so on.

**Expected results**
The printer should have changed the extruder temperature to 225 at layer 8 and kept it for the next 50 layers, until the new change kicked in.

**Additional Information**
Happy to provide extra info on request.

SecAficionado, 2019-05-15:
After re-reading the screens, it looks like it might be a matter of having more explicit labels or instructions, rather than a bug.

In the online manual at https://ultimaker.com/en/resources/20442-post-processing-plugins there is no setting for No. Layers[1]. I assumed incorrectly that the new setting(s) would be maintained for the next 50 layers and then revert to the defaults. It looks like the number of layers refers to a smooth transition over those layers, rather than my initial assumption. I would suggest changing the help text to something like:
"Enter 1 to make the change immediate, or a higher number to spread the change over the requested number of layers."

[1] Also, Cura's 3 manual completely omits the post processing plugins.

Ghostkeeper, 2019-05-17:
Thanks. I've changed the wording somewhat based on your suggestion.

commit abcac67c (2019-05-17): Improve wording of No. Layers setting

Fixes #5741.
closed 2019-05-17 by Ghostkeeper