Ultimaker/Cura #5742: [4.1 beta] Window can't be enlarged adding a custom printer

Liger0, 2019-05-15:
**Application Version**
4.1 beta

**Platform**
Win10 x64

**Printer**
Custom

**Steps to Reproduce**
Add a custom printer.

**Actual Results**
The window is very small, and you can't enlarge it or make it smaller. The drag doesn't work, though there is the icon.

**Expected results**
Guess it.

**Additional Information**
[image]

wizhippo, 2019-05-15:
I just tried on fresh install and have no issue resizing.

Liger0, 2019-05-15:
Mine was a fresh install too.

wizhippo, 2019-05-15:
My mistake.  I added then went back in to edit it.   I can confirm this does happen on the initial add print page.  You can resize when you go back in to edit.

Liger0, 2019-05-16:
Updated with another bug related to this.
If you want to modify the machine later, as the first gui when creating the machine is bugged, you'll still encounter another bug related to the gcode tabs.

Ghostkeeper, 2019-05-17:
I agree that it's inconsistent and it would be useful to allow the Add Printer window to be resized.

Can't reproduce that bug with the g-code going out of the text area though.

LipuFei, 2019-05-17:
The resizing issue is fixed in 361636af2

Liger0, 2019-05-17:
I didn't try the fix as I always wait for the releases to be published, but even if the resize is fixed, would it be possible to fix the distance between the printer and extruder tab, so that even at that minimum size, the printer settings fields don't cross the printhead settings ones?

Ghostkeeper, 2019-05-20:
Well, I can't get them to cross here. The minimum size is preventing me. I think we might have some other issues with the screen scale factor in this place.

Ghostkeeper, 2019-05-28:
Could you try this again in the 4.1 stable release that we've released today?

Liger0, 2019-05-28:
This (still) happens when resizing.
[image]

Also, the vertical and horizontal scrolling bars in the gcode don't appear, not even pointing on them. They only appear if I use the mouse wheel (is it called so? I'm not a native speaker)

closed 2019-06-06 by Ghostkeeper

Ghostkeeper, 2019-06-06:
Fixed the widths of the labels and reorganised the positioning of the two columns a bit. I hope it works on your computer as well.

Liger0, 2019-07-26:
Start and ending gcode is still out of the tab when resizing.

commit 3ea68e6c (2019-06-06): Fix width of labels in machine settings

- Elide the titles in the columns.
- Give the width of a ratio so that it doesn't matter if the font renders differently or if we resize the window.

Fixes #5742.