Ultimaker/Cura #5743: [FQ] Full screen button in printer settings dialog

yet-another-average-joe, 2019-05-15:
**Version**
all

**Platform**
Windows 10 (seems to me there is such a button in MacOS screen captures (???)

**Printer**
not related

**Steps to Reproduce**
not relevant

**Actual Results**
The panel is tiny, displaying partially the first line of the start and end G-code. 

**Expected results**
A full screen button (1 action), insteed of having to drag the window, and then resize it (2 actions)

**Additional Information**
I see that the Start G-code not updated until focus bug has been reported...

Ghostkeeper, 2019-05-17:
It should be a bit larger than you describe. For me it displays 6.5 lines of g-code in there, which is usually enough. That said, the machine settings dialogue should be resizeable. Is it not?

yet-another-average-joe, 2019-05-17:
Yes indeed it is resizable.

Many users (and myself) use many g-code lines with preheating, leveling, nozzle priming and cleaning etc. (on my side 48 lines, including some blank ones for readability)

I know nothing about QT, but in Windows API it would be the maximize button : just "WS_MAXIMIZE" to add in the "styles" with a '|' (bitwise OR) when creating the window or subclassing and registering it. Maybe it is more difficult with a multiplatform environment...

This is what I get (Windows 10), whatever version is in use :

[image]

This is much more readable, but needs :
- 1 click + drag to move the the dialog in the ussper left corner
- 1 click + drag to resize it

[image]

LipuFei, 2019-05-20:
Hi @yet-another-average-joe , The dialog is now always resizable. It's fixed in `4.1` branch by https://github.com/Ultimaker/Cura/commit/361636af29c1bd3fa55deaff2cf929b3479c2391. The fix will be in the official 4.1 release.

yet-another-average-joe, 2019-05-21:
Thanks ! I know nothing about pyQT, but I had a look to the sources and to some pyQT APIs and discussions on the Internet. It seems it is not that easy to add a maximize button to a dialog. But could a F11 key handler be added ? It seems it is easier, by far. Just an idea...

Ghostkeeper, 2019-05-22:
It shouldn't be very hard to allow this window to be maximised (just some [flags](https://doc.qt.io/qt-5/qt.html#WindowType-enum)). However with the intended use of this window as a dialogue, it makes sense to not allow it to be maximised. You are not going to be spending a significant amount of time working in this dialogue.

diegopradogesto, 2019-06-20:
@LipuFei actually in 4.1 the panel is not resizable (at least not in macOS). 

However the style has changed in 4.1, so @yet-another-average-joe could you please try in the new version and see if it looks better for you?

yet-another-average-joe, 2019-06-20:
With 4.1, the window is resizable (borders and corners). But there's a problem with the G-Code edit controls (not always updated until they get the focus, known issue).

closed 2020-06-19 by no-response[bot]