Ultimaker/Cura #5744: [FQ] More insformation beside the vertical layer slider

yet-another-average-joe, 2019-05-15:
**Application Version**
All

**Platform**
All

**Printer**
not relevant

**Steps to Reproduce**
not relevant

**Actual Results**
Only the current layer # is displayed on the left side of the Z slider.

**Expected results**
Two more informations 

- the height (thickness) of the current layer
- the Z position from the buildplate ; usefull to find glitches or holes in the original model

**Additional Information**
After slicing some models, holes can appear in some walls, due to thin walls that can't be sliced properly, but look fine in the CAD software. Having the Z position of the current layer would help a lot when fixing the original model.

Ghostkeeper, 2019-05-17:
See also https://github.com/Ultimaker/Cura/issues/5206 (not a duplicate though).

Ghostkeeper, 2019-05-21:
We discussed this and agree that it would be useful to optionally put the Z coordinate of the current layer there.

Layer thickness is not very useful though as usually all layers will just be the same, and it's not the information that you'd be adjusting with the layer view slider. We might as well show infill speed instead.

yet-another-average-joe, 2019-05-21:
Thanks !

Adapative layers is one of the greatest features ! Don't most users turn it on ? Results are amazing. Reading the chart and the model colors can be hard, and even impossible for people suffering of color blindness (dyschromatopsia)

Ok, speed and thickness are not adjusted with the slider. Maybe these informations could be displayed with an "analogic" cursor in the color charts ? Just an idea...

Off topic : I searched for the dev env you use, and there is no installation tutorial for Windows. Only for Mac and Ubuntu. And Cura can't be executed in a Virtual Box machine (due to supported GL version). Is this tutorial abandonned ?

Ghostkeeper, 2019-05-22:
The colour scale in layer view was designed to work well for people with red-green colour blindness (deuteranomaly). Other types of colour blindness are very rare, but a separate theme could be made for them that changes the colours if necessary.

> Adapative layers is one of the greatest features ! Don't most users turn it on ? Results are amazing.

About 0.2% of prints have this setting enabled. It tends to have some print quality issues like banding and pillowing. The skin thickness and support Z distance are inaccurate. I'd advise only using it if you don't use support in any case.

> And Cura can't be executed in a Virtual Box machine (due to supported GL version).

Half of the Cura team is on Windows right now. Also, one developer has VirtualBox set up to be able to test Linux-specific issues (or Windows-specific issues by seeing that they don't happen on Linux).

Hmm, I see that the tutorial on the Wiki about running from source on Windows is saying "coming soon". False promise, really. The gist of setting up on Windows is to install cura-build-environment first (instructions [here](https://github.com/Ultimaker/cura-build-environment#windows)), then compile CuraEngine with MinGW, then run cura_app.py using Python 3.5 or higher.

AbeFM, 2019-10-01:
This might be getting far afield, but the scale often makes it hard to tell what's going on visually. Often a single outlier will set the scale much wider (e.g layer thicknesses from 0.08-0.22, but the bounds of the scale are 0.08-0.4, so the subtleties are lost to the gods of dynamic range).

My real interest was the much smaller issue of adding z-coordinate to the preview https://github.com/Ultimaker/Cura/issues/6179 - I hope this wider discussion doesn't obfuscate what should be a simple change.

[image]
> Devs see CURA-6714, for our internal reference.