Ultimaker/Cura #5745: [FR] "Duplicate" button in Machine settings

yet-another-average-joe, 2019-05-16:
**Application Version**
All

**Platform**
All

**Printer**
Not relevant

**Steps to Reproduce**
Not relevant

**Actual Results**
Duplicating a printer definition is long and tedious.

**Expected results**

 A "Duplicate printer" button would be great.
Or being able to see the user's printer(s) in the "Add a printer to Cura" list

**Additional Information**
Being able to duplicate a printer definition would be usefull to people who own several indentical printers, but also for those who use one printer with several heads / nozzles and create one printer definition for each. Also be usefull when using a printer with or without a mirror depending on what to do (a few parameters have to be changed, but the start / end gcode can be long).

Imagine 1 printer, and many heads (I use 8...), swapped depending on what to do (quick swappable indexable hotends with their fans, PID params and probe Z offsets being stored in the start gcode - this works fine and will be soon on Thingiverse...). Or just several nozzles, the probe offset being known for each and updated with a M851 and M301. Creating a printer for each head/nozzle makes things much easier for the user. This means many printer definitions. Also think of the E3D tool changer (proof of concept, I know !). 

(hope you will understand what I explained...)

(I will not reopen a FR about export/import printers, but it would be a nice way to achieve such a feature...)

nallath, 2019-05-16:
Do you mean a duplicate custom FF printer? Or do you want the entire printer duplicated?

A quick workaround for you now is to save the printer in a project file and select the "new" option for the machine resolve.

yet-another-average-joe, 2019-05-16:
I was sure it will be difficult to explain (I am not 100% fluent with english, by far !)

I don't know what is a FF 3D printer :(

To make it as simple as possible, imagine the random average joe with his printer and 1 nozzle. At first he has to select a printer, or create, edit a printer profile with its start / end gcode, nozzle diameter and so on. Later , he gets 3 more nozzles and an ABL sensor. He will have to tune the z-offset for the nozzle #1 in Marlin (for example). Then, he wants to use the nozzle #2. He will re-tune the z-offset for the nozzle every time he changes the nozzle. 

Or he is lazy... He will write down the z-offset. This value depends on the machinig tolerances, but the z-offset is perfectly repeatable for each and every nozzle (I verified by measuring brims, using a micrometer). Lazyness gives ideas... He creates a 2nd printer in Cura, puts the nozzle #2 diameter in the extruder tab, and the z-offset in the start gcode with M851 + M500. Everytime the nozzle is replaced, a printer is selected. (Printer 0.4, printer 0.5, and so on).

[image]

But this is tedious. It would be easier to have a "Duplicate printer" button, and just edit the nozzle diameter and the M851 parameter everytime a new nozzle is puchased. If several preassembled heads are used, its even better for the lazy maker. Just the PID parameters to be added in the start gcode.

[image]

(for some reason, I can(t get 4.1 beta to connect to printers, but it works fine with 4.0 and SmartAvionics master ; maybe the firewall...)

It is not really about saving/restoring all the printer parameters at once in a file (this being said, it would be great !), it is about creating several configurations of the same printer depending on the nozzle, extruder, hotend, bed configuration, etc.

Maybe it is off topic, but it could help to understand the feature request Some pics of the system I developped (one indexing holder with ABL probe, and different heads, V6 and Volcano). Parameters are set when the "printer variant" is selected : nozzle diam, PID constants, z-offset. (works geat)

[image] [image]
[image]

Ghostkeeper, 2019-05-17:
> I don't know what is a FF 3D printer :(

[Fused Filament 3D printer](https://en.wikipedia.org/wiki/Fused_Filament_Fabrication#Printing), the most common type of 3D printer for plastics which works by melting a string of plastic and putting it down in the correct location. Your printer is an FF printer.

A more elegant solution would be to have swappable hotend profiles in Custom FFF Printer. We have to be aware of complexity though.

Your use case of basically building your own 3D printer was intended to be covered by adding custom machine definition files in your Cura configuration (Help -> Show Configuration Folder, then under definitions). You can add any number of extruders that can be swapped out in the same way as they can for the Ultimaker printers (except UMO and UM2). I don't think it's a common enough use case to work on support from the GUI.

yet-another-average-joe, 2019-05-17:
I have been playing with the custom definition files.

The "definitions" folder is empty. 

But I have been playing with the "definition_changes" folder, and susscessfully transfered "printers" with their gcodes from one Cura version to another, using the "MY-PRINTER-NAME_settings.inst.cfg" files. I first created vanila printers (with the same names) in Cura, and then overwrited them (I don't know if Cura scans the folder, or also maintains a list of these files !). If you confirm that it is the right way to do , this answers all the feature requests (duplicating, saving, restoring) ; It is even easier and faster than using the printers definition dialog ! (there are also the extruder definitions in the "extruder" folder)

With some googling, I found the documentation, json definition files, and anything that's needed. Under Windows, these files are in Programs/Ultimaker Cura x.x/resources

I think that this FQ could be closed ?

So FF printer = FDM printer  (where I live, we are used to say FDM)

Ghostkeeper, 2019-05-20:
> (I don't know if Cura scans the folder, or also maintains a list of these files !)

It scans the folder, so you should be fine. However a link to the file is stored in the machine instance.

> If you confirm that it is the right way to do , this answers all the feature requests (duplicating, saving, restoring) ;

I was intending to use the JSON definition files. The default ones are indeed in Program Files but you can add your own in `%APPDATA%\cura\4.0\definitions`

[FDM is a trademarked term.](https://en.wikipedia.org/wiki/Fused_filament_fabrication) That's why we decided to rename it to FFF in Cura (though under water some things are still calling it FDM).

DougJoseph, 2021-04-26:
I came here with the same question and desire. How to easily "duplicate" an existing machine, because I have more than one of the same machine, and their attributes are different just enough to want separate profiles to slice for them. 

I am on Mac, and have both 4.8 and 4.9 installed. I went into the folder on the hard drive where the machine config files are stored and tinkered with manually copying and editing, but that was giving me bad results. On 4.9 the program stopped loading. On 4.8 it loaded but with warnings.

**So, imagine my great pleasure to find that by just adding a new printer the standard way, a name brand profile for these printer makes and models was available, and it was far easier to do that way than manually fiddling with config files. Thank you, Cura.**