Ultimaker/Cura #5746: [4.1 BETA] [GUI BUG] Add a printer dialog button layout

yet-another-average-joe, 2019-05-16:
**Application Version**
4.1 BETA

**Platform**
Windows 10

**Printer**
Not relevant

**Steps to Reproduce**
Open the Add a printer dialog

**Actual Results**
The dialog can't be resized, and the last 2 printers can't be selected

**Expected results**
A resizable window with the buttons not overlaping the list box. (and maybe a sorted list : last printers are not where they should be)

**Additional Information**
screen capture

[image]

nallath, 2019-05-16:
Could you provide some more info about your display settings? The issue doesn't occur with us, so it's likely that it's specific to something in your setup.

yet-another-average-joe, 2019-05-16:
Do you mean the hardware ? My setup (4 years old, but works really well) :

Asus P9X79E-WS, Intel I7 4930K
GPUs : 3x NVidia GTX780Ti + 1x Quadro K2000
drivers : 378.66 (GTX and Quadro), newer versions make the OS unstable
(officially, the P9X79E-WS is not W10 compatible)
RAM 16GB
display :
3x Samsung Synchmaster 2233 (1680x1024, 3D, 10:9, 120Hz)
1x HP ZR22W (1920x1080, 60Hz)
3D Vision and SLI are not enabled with the desktop OS (multiboot...)
NO overclocking, CPU, RAM or GPU

[image]

I verified with 3.6 and 4.0 : no problem
4.1 beta and SmartAvionics master : problem
displaying Cura 4.1 and master on 1920p or 1680p monitors : same problem

the cursor changes on window borders, but dragging for resizing does not.

drivers :

```
nvui.dll		8.17.13.7866		NVIDIA User Experience Driver Component
nvxdsync.exe		8.17.13.7866		NVIDIA User Experience Driver Component
nvxdplcy.dll		8.17.13.7866		NVIDIA User Experience Driver Component
nvxdbat.dll		8.17.13.7866		NVIDIA User Experience Driver Component
nvxdapix.dll		8.17.13.7866		NVIDIA User Experience Driver Component
NVCPL.DLL		8.17.13.7866		NVIDIA User Experience Driver Component
nvCplUIR.dll		8.1.930.0		NVIDIA Control Panel
nvCplUI.exe		8.1.930.0		NVIDIA Control Panel
nvViTvSR.dll		6.14.13.7866		NVIDIA Video Server
nvViTvS.dll		6.14.13.7866		NVIDIA Video Server
nvWSSR.dll		6.14.13.7866		NVIDIA Workstation Server
nvWSS.dll		6.14.13.7866		NVIDIA Workstation Server
PhysX		09.16.0318		NVIDIA PhysX
NVCUDA.DLL		6.14.13.7866		NVIDIA CUDA 8.0.0 driver
nvGameSR.dll		6.14.13.7866		NVIDIA 3D Settings Server
nvGameS.dll		6.14.13.7866		NVIDIA 3D Settings Server
nvLicensingS.dll		6.14.13.7866		NVIDIA Licensing Server
NVSTVIEW.EXE		7.17.13.7866		NVIDIA 3D Vision Photo Viewer
NVSTTEST.EXE		7.17.13.7866		NVIDIA 3D Vision Test Application
NVSTRES.DLL		7.17.13.7866		NVIDIA 3D Vision Module
nvDispSR.dll		6.14.13.7866		NVIDIA Display Server
NVMCTRAY.DLL		8.17.13.7866		NVIDIA Media Center Library
nvDispS.dll		6.14.13.7866		NVIDIA Display Server

```

Ghostkeeper, 2019-05-17:
Not able to reproduce the problem here either. I wonder what pixel scale factor is being used. We should add that to the log...

yet-another-average-joe, 2019-05-17:
Bingo !!!

125%

(forgot this one...) 

[image]

So this systemmetrics is not taken into account by either Cura or QT. I use another major QT app (Subsurface divelog) at 150% (on a Surface Pro), and there is not such a problem.

For many people, high definition displays are barely readable if not scaled (HD tablet PCs, 4k monitors)

Liger0, 2019-05-19:
I have the same issue. I also can't close dropdowns for some reasons.

Ghostkeeper, 2019-05-20:
We do take the screen scale factor into account. I think the problem is that we've forgotten to apply it to one of the dimensions but not the others, so then the ratios are all out of whack.

Liger0, 2019-05-28:
Other images:
[image]
[image]

Ellecross, 2019-06-06:
Hey @Liger0 , I'm checking the latest master build today and we can resize the dialogue now. 
I've exaggerated the scale and layout of my screens and found this. But with the ability to resize the screens and dialogues, I think you can use this as workaround for now. 

This is also the case with overlapping text. 

When you say Internal dropdowns, do you mean this?  
[image]

If so, the behavior was intentionally kept the same as the previous versions of Cura. It closes when you open a different drop down.

closed 2019-06-06 by Ellecross

Liger0, 2019-06-06:
The resizing is not fluid either. After resizing the window stays in this state for around 10 seconds, and of course all the time I try to resize.
[image]

Ghostkeeper, 2019-06-07:
Can't reproduce that. It lags behind maybe 0.1s on my end.