Ultimaker/Cura #5747: [4.0.0] Post Processor Script Settings retained when changing printer but not applied to Slice

adamnock, 2019-05-16:
**Application Version**
4.0.0

**Platform**
Windows 10

**Printer**
Tevo Tarantula
Custom Printer

**Steps to Reproduce**
1. Select first printer (either)
2. Add Model
3. Select a Post Processor Script (any)
4. Change configuration of script from defaults
5. Slice and Save G-Code
6. Change Printer
7. Open Post Processor and observe settings previously set remain set
8. Slice and Save G-Code

**Actual Results**
Compare G-Code. Observe first has customised PP script settings, and second has defaults from Post Processor

**Expected results**
Both Slices should have the customised PP script settings

**Additional Information**
This was done using the Vary Temp with Height script for creating Temp Towers:
[VaryTempWithHeight.txt](https://github.com/Ultimaker/Cura/files/3187047/VaryTempWithHeight.txt)

Logs:
[cura.log](https://github.com/Ultimaker/Cura/files/3187046/cura.log)

Ghostkeeper, 2019-05-17:
The bug here is that the right-hand side of the post-processing dialogue doesn't properly refresh when switching printers. We've known about it for a while but I can't find the ticket about it any more.

adamnock, 2019-05-17:
Thats exactly it yes. My explaination might have been a bit ham fisted.
Its less of an issue once you know it happens, but it still took me a while to figure out!

nallath, 2019-05-20:
Thanks for (re) reporting this. It should be fixed for the 4.1 final release.

commit 5c33277d (2019-05-20): Correctly reset the script list on machine change

Fixes #5747
closed 2019-05-20 by nallath