Ultimaker/Cura #5751: [4.1 beta] nozzle icon out of its window

Liger0, 2019-05-18:
**Application Version**
4.1 beta

**Platform**
WIn10 x64

**Printer**
Dual extrusion

**Steps to Reproduce**
-

**Actual Results**
The section that cointains the nozzle icon isn't high enough, so when switching to the other nozzle, the unselected one has a cutted tip.

**Expected results**
Add a padding on the bottom

**Additional Information**
[image]

nallath, 2019-05-20:
That's odd, it does have the padding you would expect on my screen 
[image]

Is there some more info you could share about your screen size & DPI?

Liger0, 2019-05-20:
I use 125% window resize in Windows 10 because 100% is unplausible on QHD and higher resolutions.

Ghostkeeper, 2019-05-20:
I'm also using 125% scale on my laptop screen, but since I'm using Ubuntu the font renderer is different, and because Cura's screen scale resizing is based on font sizes, it probably causes the total height to be different here as well than on Windows.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

Liger0, 2024-11-17:
It seems fixed now.

closed 2024-11-17 by Liger0