Ultimaker/Cura #5752: Feature Request - Tree Support "Break Up Support In Chunks"

Fordari, 2019-05-18:
Hi,
I currently use Cura 4.0 and love tree supports, but on large scale prints, they are difficult to remove. It would make a world of difference if tree supports were able to be split into chunks in future versions of Cura. 
Thanks.

Ghostkeeper, 2019-05-20:
Hmm, how would it decide where to break it up? There is the danger of there just being one single branch needing to find its way down to the bottom through corners and such which becomes quite fragile.

Fordari, 2019-05-21:
Well, it could divide them similarly to the normal dividing supports up in chunks. Instead of lines or mm’s you indicate that each tree should take no more surface area than ___mm^2 . In the case that a single branch is left alone, the normal branch settings should apply to these individual branches.

nubnubbud, 2019-05-22:
I think the best way to implement it may be to group all the supports coming from (X) overhang, and stop them from touching any supports from overhang (Y), unless overhang (X) and (Y) are closer than (#)mm to each other or face a similar average direction. but this would require implementing something that can recognize an overhang as being continuous.

Ghostkeeper, 2019-05-23:
Our overhangs arrive into the tree support algorithm as polygons, so as continuous areas.

nubnubbud, 2019-05-23:
Well then, it seems the best option is to trace a line from the center of each overhang polygon toward the build plate as vertically as possible, only merging them within certain limits, then grow it outward into a tree. 

In my opinion, ideally, instead of coming to many points, the branch would start as a circle, and stay that way, rather thin, till just before the polygon, and expand outward and make an interface for the overhanging polygons to rest on.

I apologize if my language isn't very accurate, I'm still trying to build Cura so I can mess with it, so I've only seen peeks and glimpses of actual processes.

Ghostkeeper, 2019-05-23:
It's working slightly differently now, to ensure that all points get supported. It's working from top to bottom. This is the tree support algorithm, roughly:

1. In every part that needs support, spread a bunch of points (nodes) in a grid pattern.
2. Connect all nodes on the layer with their minimum spanning tree.
3. Copy all nodes to the next layer.
4. Move each copied node slightly closer towards the centre of the neighbouring nodes. This means that the leaf nodes get moved a lot towards the centre of the tree, and the rest of the nodes get a sort of relaxation.
5. Move each copied node outside of any areas that would make it eventually collide with the model.

If we want to break tree support up in chunks, we'd split up those groups of nodes that get linked together with a minimum spanning tree. They'd move towards each other automatically as you go down the layers.

Ghostkeeper, 2019-06-07:
We talked this over and agree that it would be an improvement but since tree support tends to be fairly easy to break off anyway we won't be giving this a lot of priority.

Fordari, 2019-06-07:
That’s unfortunate. I haven't had much ease with removing tree supports, especially when they wrap around the model 360 degrees. I do understand there are more pressing concerns but i would definitely enjoy seeing this added in the future!

MisterScott, 2019-12-23:
I would like to see this addressed as well.
Perhaps an approach could be 'dashes' in the support walls, e.g. add a one-line-width gap for N out of M (2 out of 3?) aligned to the axis of the branch, where the branch diameter exceeds a threshold, every <distance> around the branch.

Revolt64, 2020-03-31:
It's disappointing that this is a low priority item.

Tree supports where the base wraps 360° around the model are incredibly difficult to remove.

That situation is extremely common too, which makes this very frustrating.

zanglerecke, 2020-11-15:
furthermore it use up a lot of filament... just think a long piece with a single hole on top in the middle of it
both sides of the hole need support. 
now: support is fold around at least one edge... but there could be 2 independent trees on each side.. a lot less material used up.