Ultimaker/Cura #5754: [Feature Request] Circumference length based Minimum time

Block137, 2019-05-19:
Slow down based on time per island/circumference instead of whole layer.

This will improve quality when printing model containing multiple spikes or small pillars while not having to slow down the whole print.

The aim is to slow down when printing very small features where "Minimum layer time" failed to kick in such as when there are other areas in the same layer while not affecting original purpose of "Minimum layer time".

It's possible to enter threshold in mm of circumference but doing so will defeat the original purpose of "Minimum layer time" and will require addition of new print setting(s). So keeping time based threshold and replacing(upgrading?) "Minimum layer time" may be preferred.

smartavionics, 2019-05-19:
See https://github.com/Ultimaker/Cura/pull/3837.

Ghostkeeper, 2019-05-20:
I'll close this as duplicate of #3622.

Minimum layer time is meant to allow the material some time to solidify before putting the next layer on. I think it doesn't really apply to the slowness here because the time needed to draw a curve is more dependent on whether the material has time to flow down enough such that it attaches to the previous layer, which is a completely different ballpark.

closed 2019-05-20 by Ghostkeeper