Ultimaker/Cura #5756: [4.1 beta] More info about Advanced printer settings

Liger0, 2019-05-19:
**Application Version**
4.1 beta

**Platform**
Win10 x64

**Printer**
-

**Steps to Reproduce**
Use the printer settings plugin.

**Actual Results**
The info given about each option in the printer settings is very vague. It isn't clear, to me at least, how each option behaves on the profile.
For example, the default acceleration and jerk to me, going by plain guess, seems to be the value which tells cura how to estimate the time needed to print and doesn't change the printing behaviour at all, so it is the value used to estimate the value of acceleration and jerk when their controls are disabled.
But I have no idea how the max speed on axis and filament and the step per millimeter behave.
About the step/mm, they seem to bring the control of Estep, but in this case it isn't defined how.
There isn't a flag to enable or disable this control, and how to know if it is enabled or not, overwritting the firmware value.
A similar concept is valid for other options.
And the same is also valid for the "build volume temperature" under material. What is this?

**Expected results**
Please make it all clear on the description when pointing on the options.

**Additional Information**
[image]

fieldOfView, 2019-05-20:
>  It isn't clear, to me at least, how each option behaves on the profile.

This is largely because these settings aren't supposed to be user-accessible, but instead only for people writing printer definitions who do know how these options affect the print quality. The fact that there is a third-party plugin that makes these options accessible does not change that.

Liger0, 2019-05-20:
But I can't find any manual that explains the printer definitions too, so there still isn't a start point, and much options there are important to dial when adding a custom printer, especially given the stock printer definitions for creality machines are pretty bad.

Ghostkeeper, 2019-05-20:
Many of them are a bit hard to explain. For instance, build volume temperature eventually gets put in a `M141` command for some g-code flavours or in the g-code header for other flavours. That's too technical to explain. I've now made it slightly better though, reading it again.

For the machine settings, rather than explaining what kind of effect a setting has on the print, we've tried to explain how the definition writer can get at the value to fill in. For the speed, acceleration and jerk limits for instance this entails explaining that these are the maximum rates that the printer can reach. For many of them they are physical dimensions, such as the feeder wheel diameter. It'd be weird to explain that the feeder wheel diameter is currently only used to determine the ratio of feedrate to step size when converting to X3G formats. Such descriptions tend to get outdated fast for machine settings as well.

Liger0, 2019-05-20:
> Many of them are a bit hard to explain. For instance, build volume temperature eventually gets put in a `M141` command for some g-code flavours or in the g-code header for other flavours. That's too [...]

What about the esteps tho? They are the most unexplainable to me.

Ghostkeeper, 2019-05-20:
You mean `machine_steps_per_mm_e`?

This setting indicates the ratio between E motor steps and millimetres of extrusion, like the current description says. It's only used for conversion to X3G format. However in the future it could be used as well to get proper aliasing with rounding to steps of the feeder motor.

Liger0, 2019-05-20:
> You mean `machine_steps_per_mm_e`?
>
> This setting indicates the ratio between E motor steps and millimetres of extrusion, like the current description says. It's only used for conversion to X3G [...]

But as there is no way to enable or disable it, do they overwrite the esteps of the axis and extruder set in the firmware?

Ghostkeeper, 2019-05-20:
No, it's descriptive, not prescriptive. There always needs to be some amount of steps per millimetre so there is no way to disable it.

Liger0, 2019-05-20:
So I guess it only changes the time that Cura estimates for the print like for the default acceleration and jerk?

Ghostkeeper, 2019-05-20:
No, like I said, this setting only affects the conversion from g-code to the X3G format. This is because the X3G format stores steps rather than millimetres, as I understood.

In the future we could use this information to improve aliasing due to the rounding to step sizes in the firmware, leading to more accurate extrusion.

Liger0, 2019-05-28:
So, from  what I understand, using the esteps present the firmware in Cura would be the best thing to do.

closed 2019-05-29 by Ghostkeeper