Ultimaker/Cura #5757: missing  2 layer (one at begin, one at the end)

xvalerix, 2019-05-19:
**Application Version**
4.0 & 4.1 beta

**Platform**
Ubuntu Linux

**Printer**
Ender 3, but non important

**Steps to Reproduce**
slice with same setting with another slicer

**Actual Results**
CURA 63 layer

**Expected results**
Simplify or PrusaSlicer 2, 65 layer 

**Additional Information**
Set slicing tolerance in "Exclusive", ok with "Middle" and with "Inclusive"
Also if print in vase mode (spiralize) with 1 bottom layer, first layer is missing and do only the perimeter.

nallath, 2019-05-20:
Did you also take "Initial layer height" into consideration?  By default that's set higher than the normal layer height (0.3 vs 0.1).

xvalerix, 2019-05-20:
> Did you also take "Initial layer height" into consideration? By default that's set higher than the normal layer height (0.3 vs 0.1).

Obviously and I write it in  >"(also the first layer)"<

xvalerix, 2019-05-20:
it very easy to duplicate, get an stl cube of 20x20, slice with 0.2 layer height.
- with slicing tolerance set to "Middle" or "Inclusive" you get 100 layer (100 x 0.2)
- with slicing tolerance set to "Exclusive" you 
[image]
[image]
[image]

get 98 layer... 2 are missing

Ghostkeeper, 2019-05-20:
Yes, when setting the slicing tolerance to "Exclusive", it's intended that you get somewhat looser tolerances in the sliced (Z) direction. That's really what the setting is made for.

closed 2019-05-20 by Ghostkeeper

xvalerix, 2019-05-20:
> Yes, when setting the slicing tolerance to "Exclusive", it's intended that you get somewhat looser tolerances in the sliced (Z) direction. That's really what the setting is made for.

So missing 2 layer isn't a bug but "a features"
Ok good to know.
Any other "features"?

Ghostkeeper, 2019-05-20:
It's omitting some part of the model because you asked it to give greater tolerances.

> Any other "features"?

If you set the infill density to 0%, it's going to be missing some infill lines :stuck_out_tongue: 

Perhaps more to the point, if you set Horizontal Expansion to -0.2mm, you're going to get a similar tolerance in the X/Y direction. Very useful if you're trying to get parts to fit together but your material isn't flowing exactly as you want it to.