Ultimaker/Cura #5758: Skin Overlap Perentage Bug or Naming Issue

everycoloryouare, 2019-05-19:
**Application Version**
Cura 4.0.0

**Platform**
Windows 10 64bit

**Printer**
All

**Steps to Reproduce**
Change **Skin Overlap Percentage** with **Top Surface Skin Layers** set to non-zero value. 

**Actual Results**
Changing **Skin Overlap Percentage** only changes overlap of all Top/Bottom Layers that are not **Top Surface Skin Layers**.

**Expected results**
I would expect **Skin Overlap Percentage** to change the overlap between the Skin Layers and the walls. Either this is a bug or **Skin Overlap Percentage** should be renamed **Top/Bottom Overlap Percentage**.

**Additional Information**

Top Most Non-Skin Layer
[image]

First Skin Layer
[image]

Ghostkeeper, 2019-05-20:
Can confirm. Skin overlap doesn't seem to apply to top skin.

maukcc, 2020-04-01:
This is an annoying one. Any ETA on this?

woschtkopp, 2020-04-01:
I have no clue but I think it should be rather easy to fix.

Ghostkeeper, 2020-04-02:
I can't give you an estimate. Or rather, I shouldn't. What I can tell you is that this is currently on position 24 on our backlog. We recently re-prioritised these issues. Most likely some other bugs or features will come up before we get to it (so it won't be the 24th thing we pick up). But out of 365 open tickets in our internal work tracker it's decently high up. Some examples of bugs higher in the priority than this are:
* [Ironing offsets with half a skin line width](https://github.com/Ultimaker/Cura/issues/7036)
* [Tree support intersects with the model](https://github.com/Ultimaker/Cura/issues/5223)
* Mirror then scale results in non-uniform scaling behaviour (not reported on Github but found internally).
* [Some support areas not generated](https://github.com/Ultimaker/Cura/issues/7060).

I'd estimate this a medium-sized ticket, but that won't say anything to you. The difficulty lies in preventing the overlap to be applied twice when it's next to normal skin, according to our analysis back in May.

ScottSchmidt37, 2022-10-08:
This is still an issue in 5.1.1

Vandresc, 2022-10-21:
This is now fixed for the next release, so unfortunately still not in 5.2.

closed 2022-10-21 by Vandresc

ScottSchmidt37, 2022-10-21:
Thanks @Vandresc. I've been doing more testing and have found it quite benificial to be applying to non skin top and bottom layers as well. Was the fix to add a separate Top Surface Skin setting for it?

Vandresc, 2022-10-31:
Well, if you change the setting Skin Overlap Percentage, it should apply to all skin layers, including the top layers, that is what the desired behaviour should be. The fix is rather than the skin overlapping with the roofing we thought it made more sense for the roofing to be overlapping with the skin.