Ultimaker/Cura #5760: [4.0] Support brim printing before skirt

xelprep, 2019-05-19:
**Application Version**
4.0

**Platform**
macOS 10.14.5

**Printer**
Creality Ender 5

**Steps to Reproduce**
Configure a print to start with a skirt and enable support brims

**Actual Results**
Support brim prints first, then the skirt is printed

**Expected results**
Skirt should print first to ensure the nozzle is primed and clean

**Additional Information**
When the support brim prints first, the nozzle might not be fully primed which leads to an incomplete support brim and failed supports

Vandrasc, 2019-05-20:
Hi @xelprep , this should be already fixed for 4.1. 

If you wish, for confirmation, you could give 4.1 Beta a try, save your current Cura user directory first. 

Let us know, thanks!

Ghostkeeper, 2019-05-20:
Duplicate of https://github.com/Ultimaker/Cura/issues/4829.

closed 2019-05-20 by Ghostkeeper