Ultimaker/Cura #5764: cura 4.0 %4.1 (beta) - importing file problem

MAORUBIN, 2019-05-20:
Cura 4.0 and Cura 4.1 (Beta)

windows 10 pro

Ultimaker s5

saving a profile then uploading it

shows massage: " failed to import profile from (file location) could not find a quality type.."

nallath, 2019-05-20:
I've just tried this and it works just fine. Could you also share your log files? There should be some more hints there telling us what went wrong.

MAORUBIN, 2019-05-20:
I just created a new profile and uploaded it successfully. 

this is the file and the error massage i get from this file, were saved in the same way. 

[image]
[ABS.zip](https://github.com/Ultimaker/Cura/files/3198169/ABS.zip)

Ghostkeeper, 2019-05-20:
Yeah, I could reproduce this, both with Maorubin's ABS file but also with a file that I created myself.

The reproduce steps:
1. Switch to some configuration that has no quality profiles for it. In my case, I set it to use a 0.25mm nozzle with a 0.8mm nozzle, but it'll probably also work if you just create a custom material type or whatever.
2. Change a setting, save it to a profile.
3. Export the profile.
4. Import the profile again. This fails.

Ellecross, 2019-05-23:
I've created an internal ticket.
Ultimaker Cura devs, please see CURA-6542 :D

nallath, 2019-08-27:
This has been fixed in https://github.com/Ultimaker/Cura/commit/69e9dc13130815d7bcd1bb2ec5ae697e57083858

closed 2019-08-27 by nallath