Ultimaker/Cura #5769: [4.0.0] Strange slicing on underside hexagon

FlailAway, 2019-05-21:
**Application Version**
4.0.0  Appimage 

**Platform**
Mint 19.1, Cinnamon

**Printer**
Default

**Steps to Reproduce**
Load attached cura-horiz.obj
Use Expert mode
Select "Fine Profile
Change "Initial layer Horizontal Expansion" to "-1"
Slice
Select Preview
Rotate to see the bottom of the print
Move layer-slider to first layer

**Actual Results**
No hex until second layer, then leaves a gap.

**Expected results**
Nice hex pocket for a nut.

**Additional Information**
I had to rename the .obj file to .jpg as it would not attach .obj DUH!

I have just checked this on a different PC with a fresh download of the Appimage for both Stable and Beta versions.

Interestingly, the Beta version does an even weirder thing with the hex without even adjusting the Horizontal. It has a small hex within the bigger hex.

Hate to say it, but Slic3r and Slice3r-PE do it perfectly so pretty sure it s not the .obj file.

[image]

smartavionics, 2019-05-21:
I followed your instructions but can't see the problem. This is what I get

Layer 1

[image]

Layer 2

[image]

Bottom view

[image]

What's wrong ?

FlailAway, 2019-05-21:
Thanks for looking at it, I am getting a distinct vertical gap between the inner hex of the second and all subsequent layers and the larger first layer hex and the outer one. It looks good on your Bottom view pic, but it prints a vertical gap. 

I'll try and take a close up pic of the part. Post tomorrow.

Ghostkeeper, 2019-05-22:
When printing this, there will be a vertical gap between the inner hex of the second layer and all the subsequent layers because you set the Initial Layer Horizontal Expansion to -1, leaving a gap underneath your inner hex of 0.27mm (initial layer height). This will cause the second layer to sag, which leaves a gap above the second layer (which then causes the third layer to sag but a bit less than the second layer, and eventually it repairs itself).

I suggest to not set the initial layer horizontal expansion to -1, but to something a bit more realistic like 0.3mm.

Ghostkeeper, 2019-05-22:
I'm not really seeing much sagging while printing it.

[image]

But obviously the wall around the gap for the screw is going to sag a lot. You could consider a sacrificial bridge here.

FlailAway, 2019-05-23:
> But obviously the wall around the gap for the screw is going to sag a lot. You could consider a sacrificial bridge here.

Thanks, that's exactly what my prints are looking like. What do you mean by "sacrificial bridge?" How would I do that?

The thing is, if I use Slic3r-PE (S-PE) and set the elephant foot setting to 1, it prints as expected with no gap. I am concerned that the gap will allow the nut to turn at some point.

I do want to keep using Cura as the things you guys have done is nothing short of awesome, with Combing, simple pause control (intermediate scripts) etc. And, I'd rather not have to swap to S-PE for different things as it takes much longer for the same prints.

FlailAway, 2019-05-23:
> I suggest to not set the initial layer horizontal expansion to -1, but to something a bit more realistic like 0.3mm.

Thanks, but I am printing with PETG and at 260C and a 0.3mm elephant foot correction would not be noticed. :)

Ghostkeeper, 2019-05-28:
> What do you mean by "sacrificial bridge?" How would I do that?

A sacrificial bridge is basically that you close the circular hole for just one layer. This layer will be drawn as skin, so it will be continuous straight lines bridging the hexagonal hole. These straight lines can then (better) carry the weight of the walls on the next layer. After printing, you can cut the circular hole open with a knife or just by pushing your screw through. See an example here: https://blog.tommy.sh/posts/quick-tip-avoid-3d-printed-supports-with-sacrificial-bridge

I still don't quite understand what gap you're seeing. Maybe a picture would help?

closed 2022-01-14 by no-response[bot]