Ultimaker/Cura #5770: Dual extruder, single heater, mixing nozzle gcode order causes failure.

andymozza, 2019-05-21:
**Application Version**
4.0

**Platform**
Windows 10 and others

**Printer**
M3D Promega 
Dual extruder stepper, single mixing nozzle, single heater

**Steps to Reproduce**
When slicing for dual tools, Cura turns un-used tools or switched tools to 0 or the standby temperature, however it does this 'after' the temperature for the new tool is set.
Because this is the same shared heater, then the tool starts heading towards 0.... filament gets stuck and doesn't extrude etc.
M104 S200
G1 F600 E-2
T2
M109 S210
M104 T0 S0  <-- Set T0 to o degrees which affects the T2 which has already been switched to

**Actual Results**
Filament will no longer extrude because it's been cooled too much as the tool is switched off.

**Expected results**
M104 to turn the tool off should be sent 'before' the temperature for the new tool is set.
- very simple fix, just switch the order of the temp gcode.

M104 T0 S0 <-- Switch off un-used tool first, then work on the used tool
M104 S200
G1 F600 E-2
T2
M109 S210

**Additional Information**
I realise there are other issues related to this which have been rejected as you don't have these tools or heads etc, but I think this simple change can be visually validated and will not affect others, but be of benefit to these mixing nozzles with single shared heater.
Selection of gcode from switch between brim printed with extruder 0 and then main object printed with extruder 2 as an example.

andymozza, 2019-05-22:
You could actually put it ‘just’ before the tool switch ( T2 ) and then new temp is set after - to make sure any retraction isn’t adversely affected... however pretty sure the timing is so close it makes little difference.

smartavionics, 2019-05-22:
This problem with dual extruder shared heaters has been discussed before but the Cura developers have no interest in solving the issue as Ultimaker printers have a heater per nozzle. I am happy to work on a solution but I do not have any way of testing my code as I do not have such a printer. If you are willing to test using my Cura builds then I can help.

andymozza, 2019-05-22:
Thanks, For this change I believe it can be done simply without check boxes or UI changes ( although this is of course an option).
I still have multiple 'tools' configured to do different colour ratios, so we still need to tool switches, just need to make sure that the old tool is set to 0 or standby temp 'before' the temp on the new tool is set, so that the new temperature overrides the old temperature ( as a quick fix ).
Obviously there are more detailed changes you could do as well if you are so inclined :-)

Yes, I am willing to test yes builds.

smartavionics, 2019-05-22:
Ok, I will look into it. On the face of it, as you say, the change could be trivial but there's some complexity in the code due to the way cura schedules nozzle temp changes. I'll report back when I know more.

smartavionics, 2019-05-22:
It does look straightforward to do so can you please supply a simple project file (.3mf) that I can use to test this? Thanks. If I need a non-standard machine definition to use it, please supply that also.

andymozza, 2019-05-22:
Attached gcode output and 3mf from my machine.
Although I think you can just simulate this easily by adding more tools to any made up printer definition and setting it to use 1 tool for bed adhesion and another for the model. - as long as the other tool is not used again in the print, current cura issues the M104 TX S0 for it 'after' setting the new tool.

Search for the second T0 - it's set to S0 after the skirt is printed.  (After the temp of T2 is set).

[CFFFP_ben_floating_benchmark_kids_friendly_SWITCHESOFFTOOL.zip](https://github.com/Ultimaker/Cura/files/3206452/CFFFP_ben_floating_benchmark_kids_friendly_SWITCHESOFFTOOL.zip)

andymozza, 2019-05-22:
Also - can you check standby temperatures as well.  (not just S0)
I currently set my standby temps on each tool to match the print temp - so it causes no issue.
But if you can set the standby temp 'before' the working tool temp then this will also eliminate the need for this work around.
Again, I don't 'think' this would adversely affect other devices as the tool ID is used in those commands.

smartavionics, 2019-05-22:
Thanks for the files, unfortunately, the .3mf file only contains the model and not the profile. You need to do File -> Save rather than File -> Export. Thanks.

Ghostkeeper, 2019-05-22:
The temperature command to cool down is actually PURPOSEFULLY after the heat up command. I will not accept any change that changes this, or at least any change that changes this for printers with multiple heaters.

Initially we had the commands the other way around, but this gave a problem: If the estimated heat up and cool down speed are inaccurate, then there was a runaway effect. For instance, if the printer takes longer to heat up than expected, then the other nozzle gets more time to cool down than expected and it will be colder than expected. Then at the next nozzle switch, the nozzle will be colder than expected and thus it will take even longer to heat up and the estimate will be even farther off.

closed 2019-05-22 by Ghostkeeper

andymozza, 2019-05-22:
I can understand that for 'cool down slightly' then 're use later' then timing might make an issue, but for the completely OFF M104 TX S0 commands then this can't apply.

smartavionics, 2019-05-22:
> I can understand that for 'cool down slightly' then 're use later' then timing might make an issue, but for the completely OFF M104 TX S0 commands then this can't apply.

I agree, I will mod the code so it only swaps the order when the previous extruder is being turned completely off. You will need to ensure that all extruders have the same standby and printing temps.

andymozza, 2019-05-22:
[CFFFP_ben_floating_benchmark_kids_friendly_SWITCHESTOOLSOFF.zip](https://github.com/Ultimaker/Cura/files/3206572/CFFFP_ben_floating_benchmark_kids_friendly_SWITCHESTOOLSOFF.zip)

andymozza, 2019-05-22:
@smartavionics - thanks for continuing to investigate

Ghostkeeper, 2019-05-22:
No, for the final cooldown when an extruder is not used any more in the print it doesn't matter any more, indeed.

andymozza, 2019-05-22:
@Ghostkeeper and @smartavionics - I realise commercially for ultimaker there is nothing in it for having Cura to support these shared heater extruders.
But Cura is great - and I don't want to be forced to use 'other' slicers and there are many others who really appreciate Cura as well. Is there anything quick and dirty we can do to support them without affecting others?

There seems to be a market for these multi in - single out heads with more coming to market.
I am pretty sure a simple UI checkbox for 'extruders share the same heater' or something like this could then be used to stop sending the standby temps and extruder 'off', or re-ordering them? - unless it is end of print or pause etc.

The other issues related just seem to get closed - but no work on designing the correct solution for this.

smartavionics, 2019-05-22:
OK, I have something that can be tested. In about an hour you should find a new windows release at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

smartavionics, 2019-05-22:
I agree that having a UI setting could be introduced but, actually, I think that simply rescheduling the extruder off as we have discussed is the best solution because it doesn't add yet another setting. The user simply has to ensure that the standby and printing temps are the same and all should be fine.

andymozza, 2019-05-22:
Sorry for the delay, just ran quick test with three tools defined and I think there is another case to consider. Not sure if you can programmatically check for this easily though as not sure what the rule would be, so this may be fruitless.

I still get this in the gcode:
;TIME_ELAPSED:3850.212671
;LAYER:50
G1 F480 E1.2
G1 F17987547480000 Z11.18
T2
M104 T1 S0   **<-TOOL IS STILL SET TO OFF, BUT NO OTHER TEMPS ARE SET AFTERWARDS.**
;MESH:ben_floating_benchmark_kids_friendly.stl
G0 F7200 X353.761 Y360.611 Z11.18
G1 F17987547480000 Z10.18
G0 F7200 X351.577 Y358.853
;TYPE:PRIME-TOWER
G1 F600 E1

After layer 50, it has finished with T1 ( as it is not used again for the rest of the print), but there is no temp set for the other tool as they have both been used when printing the first 50 layers.
Is it impossible to catch this without affecting other devices ?

Not sure what you need to investigate this again, but attached the project again.
[CFFFP_ben_floating_benchmark_kids_friendly_ANDY.zip](https://github.com/Ultimaker/Cura/files/3208018/CFFFP_ben_floating_benchmark_kids_friendly_ANDY.zip)

smartavionics, 2019-05-22:
OK, I've fixed that here. It now produces this gcode...
```
;TIME_ELAPSED:3850.101630
;LAYER:50
G1 F480 E1.2
G1 F17987547480000 Z11.18
T2
M104 T1 S0
M109 S200
;MESH:Object 1
G0 F7200 X353.761 Y360.611 Z11.18
G1 F17987547480000 Z10.18
G0 F7200 X351.577 Y358.853
;TYPE:PRIME-TOWER
```

I will update my build later and will notify you here when I have done that.

Well spotted!

smartavionics, 2019-05-22:
Hi. A new dash 1 build has arrived in my dropbox. Please try it out and let me know if there are any further issues. Thanks.

andymozza, 2019-05-22:
Thanks - this build seems to work for me.
I sliced three models on the same build plate, with a skirt and circular prime tower.
Each item printed with different tool a layer at a time. Third item taller than the other two so it would have turned extruder 1 and 2 off during this print. All completed successfully.

smartavionics, 2019-05-22:
That's great, thanks for the report. If you have any further issues or suggestions, please post them on this thread. When this has had a bit more testing and is still looking good, I will submit a PR so that it can be incorporated in Ultimaker's release if they approve.

PS, now you have my build you can take advantage of some of the other groovy stuff it has!

andymozza, 2019-05-23:
Thanks - I will look into the other stuff.
I can't help thinking the cleanest solution to this is a check box on the printer definition.
" single heater multiple extruder " - or similar.
Which then just stops all tool off or tool standby temps ' except for the end of print'.
- or a shared setting between extruders if you can access the printer definition during gcode production.

Just thinking aloud, is there any issue with printers turning heater off (0) then immediately on again etc.
Also the standby temp is going to catch a lot of people out as it is separate in the UI and not intuitive that for single heater you need to make same as print temp etc.

smartavionics, 2019-05-23:
> I can't help thinking the cleanest solution to this is a check box on the printer definition.
" single heater multiple extruder " - or similar.
Which then just stops all tool off or tool standby temps ' except for the end of print'.

Yes, some time back I had a similar thought but didn't do it. I try to avoid adding settings as much as possible these days.

> Just thinking aloud, is there any issue with printers turning heater off (0) then immediately on again etc.

I wouldn't have thought it was a problem given that the printer is most likely modulating the heater power using PWM anyway. i.e. the drive voltage to the heater is being continuously switched on and off.

> Also the standby temp is going to catch a lot of people out as it is separate in the UI and not intuitive that for single heater you need to make same as print temp etc.

Maybe the help text for the standby temperature could mention that it should be the same if the heater is shared between extruders? Perhaps @Ghostkeeper , has an opinion on that.

smartavionics, 2019-05-23:
I have put a checkbox into the machine settings dialog that specifies that the heater is shared between the extruders. When this box is checked, the previous extruder temperature is not changed at all after the new extruder has been selected and its temperature set. If that is how it should behave, I can post a build later today for you to test.

[image]

Ghostkeeper, 2019-05-23:
> I try to avoid adding settings as much as possible these days.

Normally I try to avoid adding settings that nobody is going to edit or understand. Something simple, like the Z Hop Speed that came up recently, is still a good candidate for new settings.

> I wouldn't have thought it was a problem given that the printer is most likely modulating the heater power using PWM anyway. i.e. the drive voltage to the heater is being continuously switched on and off.

There is no electrical problem I don't think, but the PID regulator could be thrown off by this. Doing small temperature changes too frequently causes the actual temperature to vary wildly (~10 degrees amplitude) because the PID regulator tries to do smart stuff with the fact that there is a delay between sending electricity to the heater and the sensor heating up, and similarly for cooling down.

> Maybe the help text for the standby temperature could mention that it should be the same if the heater is shared between extruders? Perhaps @Ghostkeeper , has an opinion on that.

Yeah we could include it in the setting description. In the end, I think a better solution would be to have actual functionality for it in the machine settings, and just hide the setting based on that setting (and not use it any more in CuraEngine). Of course, we at Ultimaker can't develop that, both because we don't have the time or incentive for it and because we don't have any printer to test how it works in practice.

andymozza, 2019-05-23:
@smartavionics I 'believe' this is the behaviour that best suits. Please do a build if you can and I can test a few scenarios..... sorry brain fried a bit at the moment so having trouble thinking through all the things that might go wrong with this behaviour. Best way is just to test it through I think.

smartavionics, 2019-05-23:
Today's build provides the Shared Heater option shown above. Enabling that will stop all M104/M109 gcodes being output for extruders other than the active extruder. Also omitted are M104 gcodes for the active extruder that specify a non-zero temperature less than the current temperature. They are normally output a little while before an extruder switch to start cooling the current extruder.

andymozza, 2019-05-23:
Thanks. I will test drive tomorrow. Just one question, “Also omitted are M104 gcodes for the active extruder that specify a non-zero temperature less than the current temperature.”
- does that exclude for example the first layer being printed at 210 and rest being printed at 200 ? Or is that done with a M109

smartavionics, 2019-05-23:
That's a good point. I don't think it will get that right so that's something I will need to fix (tomorrow, I hope).

andymozza, 2019-05-24:
Maybe ignore temps changes that are not triggered in respect to any of the set print temps on that currently selected extruder?
Then for some reason if people still want a lower final print temp or a step down after first layer print temp it will still be actioned? It just shouldn’t be sent after the extruder is changed?.... would that work?

smartavionics, 2019-05-24:
Hi. I think the problem boils down to (haha) how does a shared heater system behave if the extruders don't all use the same print temperature? In the multi-heater scenario, the current extruder is pre-cooled before the switch and the next extruder is pre-heated before the switch so by the time the switch occurs it should be very close to the required temp.

The simplest approach would be to ignore the pre heating/cooling completely and simply output a M109 for the required temp at the point the extruder is switched. But that would mean that for increases in temperature the nozzle would pause at that point and for decreases in temperature, the initial portion of the zone using the new extruder would be printed hot.

A more complicated solution would issue M104s before the extruder switch so that at the point of switch the temperature would be roughly half way between the temperature of the old and new extruders. Is it really worth it?

andymozza, 2019-05-24:
For me - I use same temp for all filaments.
But I guess this needs to work for all and potentially people could load 4 filaments ( in some heads I have seen) and they print best at very slightly different temps?

andymozza, 2019-05-24:
If you are going to colour mix then you would choose filaments which are same temp. There would also be a prime tower to make sure the filament is mixed - I think also if you are printing support material and main filament then you would try to get it same temp range..... but can we guarantee that??  Like is said - for me same temp range is fine.

smartavionics, 2019-05-24:
Yes, I would expect the temps to be similar.

Anyway, I think I have something that is worth testing. It will issue M104s in advance of the extruder switch when the temps of the previous and next extruders differ. At the switch, the temp should be approx half way between the two. I will build a release for you to try.

andymozza, 2019-05-24:
The more I think about this - if you wanted to print two completely different filaments then you would by a dual nozzle setup.... and wouldn't have this problem.

smartavionics, 2019-05-24:
The new release (20190524) is now available.

andymozza, 2019-05-24:
@smartavionics Am I expecting to see the 'Shared Heater' checkbox in this one ? It doesn't show on my machine.
[image]

smartavionics, 2019-05-24:
Ah, perhaps I screwed up the Windows release. I'll check...

smartavionics, 2019-05-24:
The release looks OK, are you actually running today's build?

andymozza, 2019-05-24:
Interesting question..... I just ran the installer and it ran cura at the end. It tells me version is 'master' so it's from your build train, but how do I tell the exact version ?  ( I assumed it just overwrote the last build you gave me as it completed ). How do I find the 'exact' version form the exe running ?

andymozza, 2019-05-24:
….let me remove and re-install.

smartavionics, 2019-05-24:
When you generate some gcode, the comment at the top of the gcode should show the release date like this:

```
;Generated with Cura_SteamEngine mb-master-20190524
```

andymozza, 2019-05-24:
Thanks - Un-install / re-install worked..... for some reason I had a mixture of files from 22nd and 24th...…. maybe I had something open locking them.

andymozza, 2019-05-24:
@smartavionics I think you've cracked it.
Just running a few more different heights and tool changes to make sure.

Just notes as I go. If this is going to a wider audience ( not sure if it's possible yet ), then the default should be 'unchecked', I noticed it was already selected for me, but don't want to cause issues for ultimaker.

andymozza, 2019-05-24:
Seems to be working for me. I am running a print with three tools configured, set at 200, 205 and 210 degrees to watch the behaviour. I do see the M104, then the M109 a little later. It does pause for a few seconds in place while waiting for heatup or down as it doesn't seem to settle in time. It seems to do a z-hop while waiting for the temp. Not sure if that's reprap firmware on duet, or cura setting..... but it seems to be doing it's job well so far.

smartavionics, 2019-05-24:
Thanks for the feedback.
>Just notes as I go. If this is going to a wider audience ( not sure if it's possible yet ), then the default should be 'unchecked', I noticed it was already selected for me, but don't want to cause issues for ultimaker.

That's odd because the default value for the underlying setting is false.

>It does pause for a few seconds in place while waiting for heatup or down as it doesn't seem to settle in time. It seems to do a z-hop while waiting for the temp. Not sure if that's reprap firmware on duet, or cura setting..... but it seems to be doing it's job well so far.

There's a travel setting called z-hop after extruder switch height which (I think) adjusts that height. If I was using this feature, I think I would set it to something like 5 so to keep the nozzle well away from the print while it waits to reach the target temp.

andymozza, 2019-05-24:
might be a minor separate UI issue on the z-hop after extruder switch setting.
It actually doesn't show in the UI if Z Hop when retracted is NOT selected, BUT it still seems to be active. It was hidden which was why I couldn't see it even with all settings shown.

smartavionics, 2019-05-24:
Yeah, I noticed that also.

andymozza, 2019-05-24:
@smartavionics 5111 and 2920, 2918 as well
Not sure if you want to link them all ?

andymozza, 2019-05-24:
Where to go from here? Shall I try and get a wider audience to test?

smartavionics, 2019-05-24:
That would be useful as I can't test it. Maybe there's more folk on the Duet forum that would be willing to give it a go.

andymozza, 2019-05-24:
I'm going to try and involve the m3d discord folks as they have a dual mixing, and quad mixing head using the duet.

smartavionics, 2019-05-24:
OK, just come back here if you get any feedback or have requests for further changes. Thanks.

andymozza, 2019-05-24:
Can I just use your dropbox link ? Is this acceptable ? Also link to this gituhub so they can comment directly?

smartavionics, 2019-05-24:
The dropbox link is public, anyone can use my builds.

As for comments on this feature, it's probably actually better if we move the conversation to my repo rather than spamming the Ultimaker devs who are probably fed up by now with our discussion.

So let's abandon this issue and you can create a new one at github.com/smartavionics/Cura/issues.

gazza07, 2019-05-24:
I hope its ok but ive downloaded the latest test version and its the first time my printer has finished a dual print with out having the problems mentioned here, I'm running a Geeetech A10M with standard mixing nozzle on a duet wifi, Thank you for your time and patience looking into this problem.

smartavionics, 2019-05-24:
Hi @gazza07 , of course it's OK, thanks for the feedback. If you find any issues with the new shared heater feature or anything else that is specific to my Cura builds please open an issue at https://github.com/smartavionics/Cura/issues

gazza07, 2019-05-24:
Ok will do just gonna have a celebration coffee seeing how A10M now works with cura I really hope they allow you to add this to a release version.

walterbux, 2019-11-22:
Hello,
I am using Cura 4.4 and I have tried a print with multiple extruders with shared heater.
Result:
Heater decoupled at first extruder change and temperature going down (heater cooling).

What patch have I to use if any to circumvent the problem ?
Many thanks.

Walter

[8 more replies]