Ultimaker/Cura #5771: [4.0.0] hops Z axis on 2 corners and does not hops on other 2

HeIsTheHighlander, 2019-05-22:
**Application Version**

4.0.0

**Platform**

Win7 x64

**Printer**

Tevo Tornado

**Steps to Reproduce**

Print via USB

**Actual Results**

When printing brim of simple detail - square 24 cm with rounded corners - when head passes RR & FR corners I see Z hop which leads in defected, not round, thick, not sticked to buildplate corners. No issue on FL & RL corners.

**Expected results**

**Additional Information**

smartavionics, 2019-05-22:
Hello @HeIsTheHighlander , without seeing the model and settings it is impossible to provide any help. Please save the project file (File -> Save), and then either rename the .3mf file to .zip or zip up the .3mf file and attach the zip to this issue. Then we may have a chance to understand your issue. Thanks.

HeIsTheHighlander, 2019-05-22:
Thanks for your response. Please see the 
[CCR10_тест регулировки стола.zip](https://github.com/Ultimaker/Cura/files/3206839/CCR10_.zip)
attached file.

smartavionics, 2019-05-22:
Hello @HeIsTheHighlander , thanks for the file. I don't 100% understand your issue but I am guessing that you wish to have z-hop when the nozzle crosses the skin. Try setting the combing mode to "no skin" and see if that fixes the issue.

There is another travel setting called max comb distance without retract and I recommend you set that to something like 10 or 20.

Hope this helps.

HeIsTheHighlander, 2019-05-22:
Not exactly. Please pay attention at brim corners. Left corners of brim look good, and right corners are deformed. That's because when head passes brim's right corners, it hops up leading to such thickness which is not sticked to buildplate. I want head print right corners same as left corners, so I'm asking how to make head do not hop on brim's right corners.

[image]
[image]
[image]

smartavionics, 2019-05-22:
Ah, I understand now.

> so I'm asking how to make head do not hop on brim's right corners.

The answer is... fix your printer!

I don't think this is a Cura issue. Goodbye.

HeIsTheHighlander, 2019-05-22:
Thanks for your response. Do you think it's printer problem despite I'm printing from Cura via USB? And somehow printer prints brim with Z hop and prints part with no Z hop? It's not Cura sending this command?

smartavionics, 2019-05-22:
Sorry, I really can't see this being a Cura problem. It does look like a printer mechanical issue. Good luck!

HeIsTheHighlander, 2019-05-22:
Looks like I need to talk with someone of Cura developers. My eyes and ears clearly tell me that printer receives command to get up Z axis on that 2 corners of brim only, also my eyes told me the Z coordinate on printer's control block LCD changes accordingly.Can anyone tell me how can I contact developers?

smartavionics, 2019-05-22:
They will look at this issue and comment when they can.

I will take another look at the gcode and confirm what I said earlier.

HeIsTheHighlander, 2019-05-22:
OK, waiting for them. I can make a video how printer head moves during brim print

smartavionics, 2019-05-22:
I just sliced your project and looked for changes in Z in the gcode, there are not many there, just the few I would expect to see.
```
grep -n Z CCR10_тест\ регулировки\ стола.gcode 
7:;MINZ:0.2
10:;MAXZ:0.7
21:G1 Z15.0 F6000 ;Move the platform down 15mm
32:G0 F2100 X27.54 Y27.809 Z0.2
1927:G1 F17987547480000 Z0.7
1929:G0 F2100 X42.787 Y257.732 Z0.7
1932:G1 F17987547480000 Z0.2
8620:G0 F3150 X46.724 Y40.435 Z0.3
15300:G1 F3000 Z10
```

You can see that on line 32 it sets the z coordinate to 0.2, on lines 1927-1932 it does a little z-hop  but then z doesn't change again until line 8620 and then not again until line 15300.

Face reaiity, man, it's a printer problem.

Ghostkeeper, 2019-05-22:
When I've seen such effects on my printer it was usually that the brim was already not attaching well to the build plate and warping up a little bit, which then pushed the build plate down, making the problem worse. It's hard to get completely consistent platform adhesion in all the corners of the build plate, even for Ultimaker's printers.

HeIsTheHighlander, 2019-05-22:
Thanks guys. It's as simple as stupid. Fan duct climbed on mirror fixers. Regarding Z coordinate changes - seen that on previous issue which is solved already, my memory did bad joke on me. Looks like I need some more attention on printing process.

closed 2019-05-22 by HeIsTheHighlander