Ultimaker/Cura #5772: SO SLOW

OUPS65, 2019-05-22:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

OUPS65, 2019-05-22:
He,
4.1 beta BOOT is so so slow  with my HP spectre I7 6500 computer CPU 2500 - 64bits 8Go win 10
kind regards

fieldOfView, 2019-05-23:
Without logs, or even a mention of how slow is so so slow, there is fairly little that can be done here. Did you try other versions? Are they less slow to start?

Ghostkeeper, 2019-05-23:
I wonder what phase of the start-up is taking so much time then. Creating the interface? Loading machines?

I'm also wondering if renaming the `C:\Users\<OUPS65>\AppData\Roaming\cura` folder to `cura-backup` (while Cura is closed) helps in any way with the start-up.

OUPS65, 2019-05-23:
@fieldOfView : 4.1.0 takes 30' to be ready to use when 4.0.0 need only 17'. 
do not understand what you mean by " without logs".   connected to my ultimaker account or not do not change the delay.
@Ghostkeeper  :  both creating the interface and loading machines where slow . After  this factory reset,,  I have only created the Anycubic I3 printer...  and YES !  4.1.0 now needs only 12' .  
If I re-add a non-networked printer, with original Prusa profiles, I am still at 12' even if I import one personal  4.0.0 profiles on top of the original Prusa one.
connected to my account do not impact.
Adding "customs support", "linear advanced setting", "mesh tools", "setting guide", "version in tittel bar", "Z offset setting"  add 3' on the start.
kind regards.

PS : could be very helpfull to have the nozzel size setting displayed on the screen

Ghostkeeper, 2019-05-23:
Maybe you had a lot of printers added and that was making it slow?

> PS : could be very helpfull to have the nozzel size setting displayed on the screen

We display the active nozzle profile in the main screen, which is usually named after the nozzle size. If your printer doesn't have different nozzle profiles it displays nothing. I think that's something on the part of the maintainer of the Anycubic or Prusa definitions, whichever you were using.

trunneml, 2019-05-31:
Hi, 

Cura 4.1 is also very slow on my MS Surface Pro 4 (i5, 8GB Ram). My only Printer (Anycubic i3 Mega) is connected via OctoPrint -- which is the only extra plugin on my setup.

trunneml, 2019-05-31:
Mouse hover effects in the menu are not shown or 2 seconds to late. Typing in text fields is delayed by seconds. The problem existed directly after a fresh installation without any extra plugins.

Ghostkeeper, 2019-05-31:
Does not happen on my old MacBook Air (also i5, 8GB RAM). Cannot reproduce.

What is different about the MS Surface Pro 4 that would cause it to be slow? I would think it has an SSD as well. Perhaps the difference is Windows vs. Ubuntu?

OUPS65, 2019-06-02:
Version 4.1.0
platform : HP spectre I7 6500 computer CPU 2500 - 64bits 8Go win 10
printer : PRUSA MK3S
I re-installed Cura using the profile MK2 and not anymore  the profiles received from Prusa. Everything works now perfectly and I have impeccable prints.

about nozzel size information on the screen 
I regularly change nozzle. I would like to see on the screen what diameter is in the setting of the printer EVEN IF I only have one nozzel  on the printer :)
tks

yet-another-average-joe, 2019-06-02:
4.1 beta was slow at startup for me, but it seems the problem disapeared with 4.1 release version...

> about nozzel size information on the screen
> I regularly change nozzle. I would like to see on the screen what diameter is in the setting of the printer EVEN IF I only have one nozzel on the printer :)
> tks

just create a printer for every nozzle diameter, with the diameter being part of the name. "printer 0.3", "printer 0.4" and so on. Works fine for me. But the printer name is not displaed in Peview mode...

Ghostkeeper, 2019-06-03:
> I re-installed Cura using the profile MK2 and not anymore the profiles received from Prusa. Everything works now perfectly and I have impeccable prints.

Hmm, so you think the MK2 profiles were causing this? Where did you get those profiles?

I'm starting to think though that maybe this was caused by a plug-in that you had installed. Those are generally more powerful in what they can cause in Cura.

tkocou, 2019-06-20:
On the issue of slowness, switching from 4.0 to 4.1, a model which under 4.0 would print  1X amount of time. The very same model (using the very same printer and material settings) under 4.1 takes 3X time to print. I noticed the hot-ends would print then pause, then print, then pause, etc.
I have no idea why the slicing is so different in 4.1. Until the slicing is fixed, I will be using 4.0

Ghostkeeper, 2019-06-21:
@tkocou If you're talking about an Ender or Tevo printer then it could be that the jerk was set to be lower because people were experiencing layer shifting issues. It is intended, then. It is also a different issue than the front-end performance that is being discussed in this topic.

closed 2020-06-19 by no-response[bot]