Ultimaker/Cura #5773: Unable to connect to cloud message

askgdb, 2019-05-22:
**Application Version**
4.0.0
**Platform**
Windows 10
**Printer**
CR10s Pro
**Steps to Reproduce**
trying to open the Marketplaces it sais "Could not connect to the Cura Package Database. Please check your connection"
**Actual Results**
Marketplaces can't connect and I cannot figure out why
[image]
[image]
**Expected results**
It's a bug and there is a fix, or something with the server?

**Additional Information**
also, it pop ups a message saying "There was an error connecting to the cloud"

egore24, 2019-05-22:
Receiving same error message of "There was an error connecting to the cloud"

Ghostkeeper, 2019-05-23:
Even though this ticket was slightly earlier, I'm going to close it as a duplicate of https://github.com/Ultimaker/Cura/issues/5774 since that ticket contains somewhat more useful information as to connection debugging. Please continue the conversation there!

closed 2019-05-23 by Ghostkeeper