Ultimaker/Cura #5774: DNS Resolution does not work for api.ultimaker.com

zepich, 2019-05-22:
**Application Version**
4.0

**Platform**
Windows 10 64bit

**Printer**
Ultimaker 3

**Steps to Reproduce**
- Execute "ping api.ultimaker.com" in a command line window

**Actual Results**
It does not work since it can't resolve 'api.ultimaker.com'. This means that the cloud functionalities in Cura 4 and in the printer are not working correctly.

**Expected results**
Resolving 'api.ultimaker.com' and working cloud functionality

**Additional Information**
If you use the Google DNS server it will work, but if you use some other DNS provider like Quad9 it does not work.

Command:
`dig A api.ultimaker.com @8.8.8.8`

Result:
`api.ultimaker.com.      299     IN      A       35.240.98.38`

Command:
`dig A api.ultimaker.com @9.9.9.9`

Result:
No resolution

You can even try some online DNS propagation tools: https://dnschecker.org/#A/api.ultimaker.com

You'll see the completely different results worldwide.

You can also try to resolve **ultimaker.com** and you'll see strange results:

https://dnschecker.org/#A/ultimaker.com

ChrisTerBeke, 2019-05-22:
We're having issues with our DNS provider and are actively working on resolving them. Please hang in there!

ChrisTerBeke, 2019-05-23:
The DNS issues have been resolved, so everything should be up and running again. Apologies for the inconvenience.

closed 2019-05-23 by ChrisTerBeke