Ultimaker/Cura #5779: [4.1.0 Beta] Machine settings Start G-code input display corrupt

bobstro, 2019-05-24:
**Application Version**
4.1.0 Beta

**Platform**
MacOS Mojave 10.14.5 on 27" iMac 5K

**Printer**
Prusa i3 Mk3

**Steps to Reproduce**
Attempt to modify printer Start G-code

**Actual Results**
Display of gcode is corrupted. Scroll bars are erratic and appear in middle of dialog box. Pasted text is accepted and works, but it is very difficult to review.

**Expected results**
Entry and editing of text using normal platform editing commands, proper display of text.

[XYZ Cube.3mf.zip](https://github.com/Ultimaker/Cura/files/3218829/XYZ.Cube.3mf.zip)
[image]

**Additional Information**

Liger0, 2019-05-25:
Maybe related: #5742

Ghostkeeper, 2019-05-28:
Yeah I think it's a duplicate of #5742. Please continue discussion there.

closed 2019-05-28 by Ghostkeeper