Ultimaker/Cura #5780: CLI mode: stdout and stderr not visible

thopiekar, 2019-05-25:
**Application Version**
4.0.0

**Platform**
Windows

**Printer**
Doesn't matter

**Steps to Reproduce**
Running "CuraCLI.exe --help" in PowerShell

**Actual Results**
No output

**Expected results**
Output as I implemented it about one year before (Cura 3.2):
[image]
The beginning was noisy as it gave debug info from the initialization, but I was able to see the parser's output at the end. The plugin I'm working on needs to add command line options and I want to see the output directly instead of wasting time in looking for the stdout and stderr file.

**Additional Information**
Please bring this functionality back. The stdout and stderr files are not useful during development as getting direct feedback from Cura via command line. 
I have the suspect that this functionality has been removed during a cleanup months before.

thopiekar, 2019-05-25:
Additionally, I noticed that I can't register command line options anymore.
Have the feeling that https://github.com/Ultimaker/Uranium/pull/491 will fix it, but haven't tested it.

Again, please bring this functionality back. I need this to continue on my project.

thopiekar, 2019-05-28:
@Ghostkeeper Can you give me an update on whether this problem is somewhere on your board? Or not?

Thanks 😉 👍

Ghostkeeper, 2019-05-29:
Yeah we discussed it this morning and added it (CURA-6549).

thopiekar, 2019-05-30:
Is there any statement you can give regarding priority?
Do you need the current state of the plugin which makes use of this functionality, so you have something to test?
If there is anything I can do, please let me know!

Ghostkeeper, 2019-05-31:
As discussed this morning in person when you came to visit us, we didn't set a priority on this yet. If nothing is done, the priority will be low. I'll leave it up to @diegopradogesto and his stakeholders to decide on the priority.

closed 2024-12-07 by github-actions[bot]