Ultimaker/Cura #5781: [4.1.0 Beta] Z-hop option uses unreasonably slow speed in g-code

extesy, 2019-05-26:
**Application Version**
4.1.0-BETA

**Platform**
Windows 10 x64

**Printer**
Wanhao i3 Duplicator Plus Mark II

**Steps to Reproduce**
Configured Z-Hop options:
1. Z Hop When Retracted: yes
2. Z Hop Only Over Printer Parts: yes
3. Z Hop Height: 0.6mm

**Actual Results**
Fragment of a generated g-code:
```
G1 F2400 E2675.23268
G1 F17987547480000 Z17.9
G0 F7200 X105.229 Y81.446 Z17.9
G0 X114.621 Y65.828
```
As you can see, the rate is so large that it essentially inserts a very long pause into the printing process.

**Expected results**
Normal rate is used for z-hop.

baxi69, 2019-05-26:
Ah ok, I just wanted to send a video with this problem. :D
Yes this is very nasty, slows print endless down.
Thx for finding this bug on the point.
Cura is kind of non-usable with this bug.

baxi69, 2019-05-28:
Was this fixed in the currently released stable version?

Ghostkeeper, 2019-05-28:
Duplicate of https://github.com/Ultimaker/Cura/issues/5753

Not yet fixed, no.

closed 2019-05-28 by Ghostkeeper

keeeal, 2019-07-03:
This is still not fixed. I could not find the duplicate issue so I am posting here.