Ultimaker/Cura #5782: [4.0.0] Needs Retraction + wish-list

FlailAway, 2019-05-26:
**Application Version**
4.0.0

**Platform**
Mint 19.1, Cinnamon

**Printer**
Any I am guessing

**Steps to Reproduce**
Print with opening.

**Actual Results**
Snotty tracks between end/start points

**Hoped for results**
No snotty tracks

**Additional Information**
1: Bug?? Or, is there an option for retraction when scooting from one place to another on the same layer? That snotty trail echos through each layer.

2: Angles: I had to rotate the part slightly to miss some stuff and forgot to add that angle to the Infill setting of 45-deg. Is there an option to lock the Infill angle to the relative angle? When I rotate the part the 7-deg, the Infill-45 would be adjusted to 52-deg.

3: First layer only, infill first: If a part has a bunch of small holes, it would be better to print the infill first on the Initial Layer, then print the holes. I know I can set that for the entire print but I find I get a better finish on the completed print if the Infill is printed after the perimeter. But sometimes those lonely little circles don't seem to stick well on that first layer.

[image]

zebons, 2019-05-28:
For point 1 : do you put the Combing mode to Not in infill (or even more deactivate it) ?

Ghostkeeper, 2019-05-28:
> 1: Bug?? Or, is there an option for retraction when scooting from one place to another on the same layer? That snotty trail echos through each layer.

As Zebons said, the different combing mode are able to accommodate your preference in this regard. When should it try to avoid walls and when should it just go over them with a retracted move?

I'm going to close this issue since this has already been implemented and it's the "main issue". If you need to, please create separate tickets for separate issues. There is no option in Github for "1/3rd of the issues in this issue are fixed".

> 2: Angles: I had to rotate the part slightly to miss some stuff and forgot to add that angle to the Infill setting of 45-deg. Is there an option to lock the Infill angle to the relative angle? When I rotate the part the 7-deg, the Infill-45 would be adjusted to 52-deg.

This is pretty much impossible since not all rotations are rotations around the Z coordinate. For instance, if you were to rotate a model 180 degrees around the X axis and 180 degrees around the Y axis that would be equal to a rotation of 180 degrees around the Z axis, but you've never actually touched the blue circle for rotation around the Z axis. On top of that, some patterns are 3D while others are 2D.

> 3: First layer only, infill first: If a part has a bunch of small holes, it would be better to print the infill first on the Initial Layer, then print the holes. I know I can set that for the entire [...]

One possibility to achieve this is to use a cutting mesh ("Modify Settings for Overlap with Other Models") that spans the first layer and change this setting per-object on the cutting mesh. The request to add this as a normal setting is valid though, for ease of use.

closed 2019-05-28 by Ghostkeeper