Ultimaker/Cura #5784: [4.1.0]X and Y offsets fields do not allow negative values

bellzw, 2019-05-27:
**Application Version**

4.1.0 beta
**Platform**

PC Windows 10
**Printer**

bibo2 dual (as supplied with installation in bibo2_dual.def.json)
Also occurs with json files supplied by manufacturer
Also occurs with Creality CR X definition.
**Steps to Reproduce**

Open 4.1.0 beta
Add printer->other->bibo2 dual->Add
Machine settings screen appears
Select tab Extruder 2
Enter -0.02 in Nozzle x offset text box

**Actual Results**

Minus sign does not appear.  It is not at all obvious if it was accepted since no error messages appear.
**Expected results**

Should see -0.02 in Nozzle x offset text box
**Additional Information**

Same behavior when entering a negative number in Nozzle y offset.

Ghostkeeper, 2019-05-28:
Oops, it was indeed not allowing negative values. The `-` wasn't getting typed at all and therefore also wasn't used to slice with.

Fixed for 4.2.

closed 2019-05-28 by Ghostkeeper

Tomaskom-cz, 2019-06-10:
I'm on Linux and had the issue as well. Just commenting to make sure it gets fixed for all platforms. ;)

I was able to circumvent the issue by manually editing the file
/home/Tom/.local/share/cura/4.1/definition_changes/creality_ender3_extruder_0+%232_settings.inst.cfg by overriding the value machine_nozzle_offset_y
This then shows up in the UI too and is used in gcode generation. (But once you delete the minus sign, you can't type it in again of course).

The problem is very likely related to UI change between 4.0 and 4.1, where Cura (at least on Linux) started using it's own gpraphics style dialog for Machine Settings, instead of the more "system native" one before.

Ghostkeeper, 2019-06-11:
I work on Linux myself and the change works for me.

Foxsa, 2019-07-12:
Thanks a lot! 

Actually I work on Windows 10 and for me file is located here: 
C:\Users\<username>\AppData\Roaming\cura\4.1\definition_changes

commit f6627daa (2019-05-28): Allow negative values for nozzle offsets and print head shape

That's half of the point of these settings, anyway.

Fixes #5784.