Ultimaker/Cura #5786: factor K in parametre materiaux

sim1953, 2019-05-27:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

sim1953, 2019-05-27:
Hello,
it would be nice to be able to enter the LIN_ADVANCE K-factor in the material parameters to make the prints more accurate

zebons, 2019-05-27:
it's possible via the MaterialSettingsplugin (https://github.com/Ultimaker/Cura/issues/3946).
However I encounter an issue here because after adding this parameter at the material profile, I cann't enter a decimal value here. I opened an issue  #5787

Ghostkeeper, 2019-05-28:
I'm open to integrating this factor in Cura, if done in a neater way (i.e. either CuraEngine or a post-processing script). However we can't expect Ultimaker to spend any time on it since their printers don't understand this g-code. And me personally I also can't implement it since I have no printer that understands the g-code either to test with.

Until someone makes a PR to integrate this, I suggest you report this at the repository of the maintainer of those plug-ins: https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/issues
Honestly, he's probably listening in here, but for his visibility and Cura's cleanliness, it's better to track these issues at the appropriate repositories.

closed 2019-05-28 by Ghostkeeper