Ultimaker/Cura #5787: [4.0] Decimal number not possible for Linear advance factor at Material profile level

zebons, 2019-05-27:
I installed the LinearAdvanceSettingplugin to manage Linear Advance parameter. This plugin adds a new field in the Material section named Linear Advance Factor. In the field I can choose my optimum K factor as decimal number.
As this number is per material, I decide to store this parameter at the material profile level via the MaterialSettingPlugin. With this plugin, I can add the new parameter in the material profile.
However it's not possible to enter here a decimal number.
Can you take a look on that issue ?

fieldOfView, 2019-05-27:
The problem is that the Material preferences pane uses a very naive way to decide whether to display/allow decimals. It literally only shows decimals for settings that have a "mm" unit. This would need to be changed. I have considered monkey-patching the Materials preferences pane in the mentioned Material Settings plugin (https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/), but I have other priorities at the moment.

zebons, 2019-05-27:
Thanks for your response. Linear advance factor is directly link with the filament used so this parameter should be stored at material level. It'd be possible with the plugin MaterialSettingsPlugin but unuseable due to that issue. 
With latest Marlin firmware and linear advance enhancements, this behaviour will be asked soon by more and more users

fieldOfView, 2019-05-28:
> With latest Marlin firmware and linear advance enhancements, this behaviour will be asked soon by more and more users

Not by Ultimaker users that are paying the bills though, because Ultimaker printers don't support Linear Advance. I'm not saying this will never get fixed, but I don't think it will have a high priority.

zebons, 2019-05-28:
Thanks,
Can I propose an alternative and temporary solution ?
As the Linear Advance factor parameter is only used for putting it in the start G-code command and isn’t used at all by Cura to define it in mm avoiding decimal values in the MaterialSettingsPlugin ?
Of course it isn’t an optimal and definitive solution but could help us to use the Linear Advance concept more easely (actually we have to dupplicate our material profiles for each filament having a different K factor).

Is there a way to do it locally ?

zebons, 2019-05-28:
As the Linear Advance factor parameter is only used for puting it in the start G-code command and isn’t used at all by Cura, is it possible to define it in mm permiting decimal values with the MaterialSettingsPlugin ? => of course not avoiding but permiting decimal values

De : Frédéric NEE<mailto:zebons@hotmail.com>
Envoyé le :mardi 28 mai 2019 08:54
Objet :RE: [Ultimaker/Cura] [4.0.0] Decimal number not possible for Linear advance factor at Material profile level (#5787)

Thanks,
Can I propose an alternative and temporary solution ?
As the Linear Advance factor parameter is only used for putting it in the start G-code command and isn’t used at all by Cura to define it in mm avoiding decimal values in the MaterialSettingsPlugin ?
Of course it isn’t an optimal and definitive solution but could help us to use the Linear Advance concept more easely (actually we have to dupplicate our material profiles for each filament having a different K factor).

Is there a way to do it locally ?

zebons, 2019-05-28:
That alternative solution’d only impact the LinearAdvancePlugin (change the definition of the new parameter Linear Advance Factor and define it in millimeters) and will disappear when Cura’ll integer that parameter in the master branch (in that case the plugin LinearAdvance’ll not be usefull)

________________________________
De : Frédéric NEE
Envoyé : Tuesday, May 28, 2019 9:03:43 AM
À : Ultimaker/Cura
Objet : RE: [Ultimaker/Cura] [4.0.0] Decimal number not possible for Linear advance factor at Material profile level (#5787)

As the Linear Advance factor parameter is only used for puting it in the start G-code command and isn’t used at all by Cura, is it possible to define it in mm permiting decimal values with the MaterialSettingsPlugin ? => of course not avoiding but permiting decimal values

De : Frédéric NEE<mailto:zebons@hotmail.com>
Envoyé le :mardi 28 mai 2019 08:54
Objet :RE: [Ultimaker/Cura] [4.0.0] Decimal number not possible for Linear advance factor at Material profile level (#5787)

Thanks,
Can I propose an alternative and temporary solution ?
As the Linear Advance factor parameter is only used for putting it in the start G-code command and isn’t used at all by Cura to define it in mm avoiding decimal values in the MaterialSettingsPlugin ?
Of course it isn’t an optimal and definitive solution but could help us to use the Linear Advance concept more easely (actually we have to dupplicate our material profiles for each filament having a different K factor).

Is there a way to do it locally ?

Ghostkeeper, 2019-05-28:
> Note that the naive implementation works for the intended usecases as designed by Ultimaker, so I don't think fixing this properly will have a high priority for the Cura dev team.

If it's a change I can make in 2 minutes and the change is obvious/safe, I'm willing to help out. It's illogical that it depends on the unit anyway.

fieldOfView, 2019-05-28:
> If it's a change I can make in 2 minutes and the change is obvious/safe (...)

The proper way to fix it would be to make fields that behave like SettingTextField.qml from the sidebar, but look like system native fields. But that would take more than 2 minutes and since you are moving away from the system-native looks and plan to eventually replace the preferences with a "Cura-themed" version, that will be wasted effort.

zebons, 2019-05-28:
@Ghostkeeper agree with you totaly illogical. Types of fields'd be taken from the data dictionnary not issued from the unit !
You really can do a change for that ? It'd be great !

fieldOfView, 2019-07-12:
The Material Settings plugin is now available from the Marketplace. If you install it, it will replace the Print settings tab to show Cura-styled widgets that take the type of setting as well as valid min/max values into account. It does this for all settings, not just the Linear Advance factor setting.

fieldOfView, 2019-07-12:
I think this issue can be closed, unless you want to keep it open as a note that the Material settings fields on the Materials pane of the preferences need better widgets when the preferences dialog is reworked.

zebons, 2019-07-13:
I install the plugin from Market Place and indeed, we now can manage decimal parameters via Material profile. Thanks a lot !
I'll close this issue.

closed 2019-07-13 by zebons