Ultimaker/Cura #5788: [4.1 Beta] A layer is moved from the middle to the bottom of the model

extesy, 2019-05-27:
**Application Version**
4.1-BETA

**Platform**
Windows 10 x64

**Printer**
Wanhao i3 Duplicator Plus Mark II

**Steps to Reproduce**
Slice the attached model.

**Actual Results**

Layer 70 is missing in the middle of the model:

[image]
[image]

In fact, it gets moved to the bottom of the model (below layer 1).

[image]

Model and generated g-code: [pencil-case.zip](https://github.com/Ultimaker/Cura/files/3224449/pencil-case.zip)

**Expected results**
Normal slicing with layer 70 in its normal position.

fieldOfView, 2019-05-27:
Duplicate of https://github.com/Ultimaker/Cura/issues/3815
This is a sporadic visualisation bug in Layer view. The gcode will print correctly. If you slice the model again, the layer will also be placed correctly.

Liger0, 2019-05-27:
>  If you slice the model again, the layer will also be placed correctly.
No, it won't.

extesy, 2019-05-27:
@fieldOfView In #3815 there was no g-code. In my case, I did save the g-code - maybe that will help with debugging?

fieldOfView, 2019-05-28:
When I load the gcode into Cura (4.1.0 beta), all the layers are in the correct order. There is nothing wrong with the gcode, it will print as expected. It still looks like a duplicate of #3815 to me.

Liger0, 2019-05-28:
Yeah it is for sure, but reslicing wil lstill display the layer in the wrong position. I have this bug frequently and even if I restart the PC and reload the model I still have it until I change some settings (ore more than one, sometimes one is not enough) in the profile

nallath, 2019-05-28:
And provide the reproduction steps, because that's what's stopping us from fixing it.

nallath, 2021-07-07:
Closing this as a duplicate in favor of https://github.com/Ultimaker/Cura/issues/7412

closed 2021-07-07 by nallath