Ultimaker/Cura #5789: [4.0.0] Text corruption on Monitor view on Mojave

martynbrearley, 2019-05-27:
**Application Version**
4.0.0

**Platform**
MacOS 10.14.5 (Mojave)
MacBook Pro Retina 15" Mid 2014

**Printer**
S5

**Steps to Reproduce**
Frequently occurs when trying to monitor a print in progress, after creating the job and swipe-switching desktops a few times.
Only seems to happen once a network print has been started.
Closing Cura app and restarting it usually clears the issue.

**Actual Results**
Text becomes heavily corrupted

**Expected results**
Text should remain readable

**Additional Information**
[image]

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

martynbrearley, 2024-11-23:
Thanks for checking Greg. It does appear to have been fixed a while back. I've currently got 5.7.2 and it doesn't occur there. I'm on Sonoma 14.7 so it could have been MacOS that changed. Hard to say.
Please go ahead and close.

GregValiant, 2024-11-23:
Thanks.  I'll close it.

closed 2024-11-23 by GregValiant