Ultimaker/Cura #5790: Writing a Cura Plugin / adding new Settings with cildren is not working for me

TimoErdmann, 2019-05-27:
Hey
I have to write my own plugin and i am trying to own settings with some subsettings to the settings container. I am able to add one setting.

But I need more than one setting and need the option to enable and disable this plugin in the Print Settings.

```
self._setting_dict = {
            "label": "Enable my special plugin with childs",
            "description": "Enbales my special plugin!",
            "type": "bool",
            "default_value": 0,
            "settable_per_mesh": False,
            "settable_per_extruder": False,
            "settable_per_meshgroup": False,
            "children":
            {
            "experimental_my_plugin_float_setting":
                {
                    "label": "Diameter",
                    "description": "Adjusts the diameter of the filament used. Match this value with the diameter of the used filament.",
                    "unit": "mm",
                    "type": "float",
                    "default_value": 2.85,
                    "minimum_value": "0.0001",
                    "minimum_value_warning": "0.4",
                    "maximum_value_warning": "3.5",
                    "settable_per_mesh": False,
                    "settable_per_extruder": True
                }
            }
        }
```
 
I added a "children" to the dict like it is in the FDMPrinter.json.def

The "Setting Visibility" shows the childs, but they are not shown in the Print settings. 

You can download my test plugin here: https://drive.google.com/drive/folders/15o8DGH0I47gQZc3c8LiP2Wh-YlmXRimc?usp=sharing

Greetings

fieldOfView, 2019-05-28:
Continued from https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin/issues/8

Adding a setting like you do in this plugin is not officially supported and "hacky". I think you may have to add the settings one at a time, one as a child of the category, and one as a child of the first setting you added.

Ghostkeeper, 2019-05-28:
Adding a setting like that on the fly is probably best done by adding a new definition to the fdmprinter definition container.

Maybe it has something to do with the child setting being per-extruder while the parent setting is not. Keep in mind that in Cura, child settings always override their parent settings for more specific cases. Also child settings may still be visible even if their parent is not. Your use case is something similar to the support settings. Cura has the support settings completely separate (not as children), and hides/shows settings based on the `enabled` property:

    "enabled": "support_enable",

I suggest that you try something similar.

closed 2019-05-28 by Ghostkeeper