Ultimaker/Cura #5792: Prime Tower Brim self-intersections

BagelOrb, 2019-05-28:
**Application Version**
4.1 beta

**Platform**
Windows 10

**Printer**
UM3

**Steps to Reproduce**
Enable brim for prime tower
put object close to prime tower

**Actual Results**
Toolpaths for prime tower intersect toolpaths for model

**Expected results**
There should never be multiple toolpaths in the same location.
The areas of print features should never be overlapping (except for as much as the infill overlap etc.)

**Additional Information**
[image]

BagelOrb, 2019-05-28:
Also happens in 4.1 final, which was just posted online:
[image]

BagelOrb, 2019-05-28:
[tower_brim_bugs.3mf.rename.zip](https://github.com/Ultimaker/Cura/files/3228010/tower_brim_bugs.3mf.rename.zip)

rburema, 2019-05-29:
Thanks for the report: See https://github.com/Ultimaker/Cura/pull/5806

Liger0, 2019-05-30:
It would also be useful to calculate a working position for the purge tower.
Normally, when enabling it, the slicing will fail because the purge tower by default is over the build plate.

BagelOrb, 2019-05-31:
@Liger0 that's a wholly different issue. For Ultimaker machines the tower *has* to be on the build plate.

Ghostkeeper, 2019-06-03:
Fixed for the next release. Thanks for the report!

closed 2019-06-03 by Ghostkeeper

rburema, 2019-06-04:
That depends on what you mean by the brim-width issue? If you mean that the brim becomes double-wide if both 'brim for prime tower' is selected _and_ the 'normal' adhesion method is also set to brim, then it's expected behaviour.

Ghostkeeper, 2019-06-04:
No, the fact that the brim is wider than the disallowed area of the prime tower wasn't changed because it's not a bug.

Like the brim around printed models, the brim around the prime tower brim can intersect with other brims. They will be unioned then, so models are allowed to be placed close to the prime tower.

BagelOrb, 2019-06-04:
> If you mean that the brim becomes double-wide if both 'brim for prime tower' is selected and the 'normal' adhesion method is also set to brim, then it's expected behaviour.

No I didn't mean that, but that sounds like a bug which is easily fixed, just by subtracting the normal brim width from the user specified towre brim width when creating the tower.

BagelOrb, 2019-06-04:
Never mind the original (second) issue. I was confused that the tower was smaller than the disallowed area, while the tower brim was larger than the disallowed area.

BagelOrb, 2019-06-04:
Why not fix the double brim issue, though?

rburema, 2019-06-04:
Why? It can be a feature as well. Removing it would limit the ammount of different behaviour you can get, and the double brim may be useful in certain contexts. Plus may be sort of expected depending how you interpret both settings being present/on.

rburema, 2019-06-04:
It is different, since right now the prime-tower brim is just an on or off thing, and you can't specify the brim-width specifically for prime-tower right now, which means that, if you want a larger brim for prime tower than the rest of the model, you can.

I see your point though, I'll bring it up during the CCB.