Ultimaker/Cura #5793: 4.1b prepends heating to gcode

thesec, 2019-05-28:
Following 
https://github.com/Ultimaker/Cura/issues/2041

I have tried both
```
M109 S{material_print_temperature}
```
and
```
M109 S{material_print_temperature_layer_0}
```
with the extruder start code.

Yet CURA keeps prepending

```
M104 S...
M105
M109 S...
```

What am I missing that will have it stop prepending it?

fieldOfView, 2019-05-28:
Please post more of your start gcode and the resulting gcode.

Ghostkeeper, 2019-05-28:
I'm unable to reproduce this.

I have this start g-code in Cura:
```
;start g-code
M109 S{material_print_temperature_layer_0}
;fin
```

It's producing the following start g-code for me:
```
;FLAVOR:Marlin
;TIME:1633
;Filament used: 0.420144m
;Layer height: 0.1
;MINX:32.202
;MINY:32.202
;MINZ:0.3
;MAXX:67.798
;MAXY:67.798
;MAXZ:20
;Generated with Cura_SteamEngine master
M82 ;absolute extrusion mode
;start g-code
M109 S200
;fin
M141 S35
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:198
;LAYER:0
M107
G0 F3600 X37.501 Y37.329 Z0.3
;TYPE:SKIRT
G1 F1500 E0
G1 F1800 X37.781 Y37.058 E0.00733
```

Project file: [overwrite_start_gcode_temperature.3mf.rename.zip](https://github.com/Ultimaker/Cura/files/3228704/overwrite_start_gcode_temperature.3mf.rename.zip)

thesec, 2019-05-28:
Am having the issue with 

```
T0 ;switch to the left extruder
M109 T0 S{material_print_temperature_layer_0} ;wait for the hotend to heat
G92 E0 ;zero the extruded length
G1 F72 E15 ;extrude 15mm of feed stock
G92 E0
G4 P2000 ;stabilize hotend's pressure
```
as start code on the first extruder, set in the extruder's def.json
as 
`"machine_extruder_start_code": { "default_value": "T0 ;switch to the left extruder\nM109 T0 S{material_print_temperature_layer_0} ;wait for the hotend to heat\nG92 E0 ;zero the extruded length\nG1 F72 E15 ;extrude 15mm of feed stock\nG92 E0\nG4 P2000 ;stabilize hotend's pressure" },`

and

```
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z5 F200 ;safety Z axis movement
M190 S{material_bed_temperature_layer_0} ;wait for the bed to heat
```
set as start code in the printer's def.json:
```
{
    "version": 2,
    "inherits": "fdmprinter",
...
    "overrides": {
        "machine_start_gcode": { "default_value": "G21 ;metric values\nG90 ;absolute positioning\nM107 ;start with the fan off\nG28 X0 Y0 ;move X/Y to min endstops\nG28 Z0 ;move Z to min endstops\nG1 Z5 F200 ;safety Z axis movement\nM190 S{material_bed_temperature_layer_0} ;wait for the bed to heat" },

```

fieldOfView, 2019-05-28:
The extruder start gcode is only included upon the first switch, so not at start. Therefore, Cura only checks the start gcode for a custom heatup sequence. In other words, {material_print_temperature_layer_0} needs to be in the start gcode (not the extruder start gcode), even if in a comment.

Ghostkeeper, 2019-05-29:
Yeah, this is expected behaviour.

The extruder's start g-code is not executed at the first run because some printers need to heat up their extruders differently for extruder switches than for the start of the print.

thesec, 2019-05-29:
This is illogical, undocumented - at least with #2041 - and uncorrect as the extruder start code is very well rendered to the gcode, right after the printer's.

Unless I get "included upon the first switch, so not at start" wrong this would then be a bug? Otherwise it is easily improved by adding a listener to the extruder's start code where someone would logically put the extruder's gcode at.

Ghostkeeper, 2019-05-29:
In #2041 extruder start g-code isn't mentioned. The g-code you're seeing is part of their start g-code. The extruder start g-code and end g-code has always been output only on extruder switches. I hope we can find a way to document that better then.

The extruder start g-code can't automatically be added to the printer start g-code (or output there). This is because the behaviour of an extruder switch needs to be different from the behaviour of starting a print for some printers. For instance, you need to be retracted to a different position during stand-by (so you need to unretract a different distance), or you need to move to a certain location on the build plate to an extruder switching bay. You don't want this g-code to be executed upon the start of a print as well.

thesec, 2019-05-29:
I am not saying or wanting the extruder's start g-code to be added to the printer's.

I am saying that, looking at my g-code, the extruder's start code is being added to the output / saved g-code just after the printer's:

```
;Generated with Cura_SteamEngine 4.1.0-BETA
... cura's start code  ...
... printer's start code ...
... extruder's start code ...
;LAYER_COUNT:42
;LAYER:0
...
```

Again: if the extruder's start code is as not expected there ("You don't want this g-code to be executed upon the start of a print as well.") ... this is a bug, isn't it?

On the other hand: where do I find documentation on how to decide on the extruder to heat up in the printer's start code? Do you support 'if' clauses or the like?

Ghostkeeper, 2019-05-31:
Yeah, that is a bug then. It shouldn't execute the extruder's start g-code there. Maybe you can provide a project file that reproduces that issue?

> On the other hand: where do I find documentation on how to decide on the extruder to heat up in the printer's start code? Do you support 'if' clauses or the like?

[Here's](https://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature) some documentation on how to command certain extruders to heat up.

The GUI doesn't support logic expressions. This can be achieved by profile developers though in the `"value"` property of the setting. [Here](https://github.com/Ultimaker/Cura/blob/95e7835da80d2f945ba258171eef953f8ad7ffec/resources/definitions/Mark2_for_Ultimaker2.def.json#L149) is an example of a printer definition that does this.

thesec, 2019-06-03:
Thank you, will dive into the profile then to prevent two heated nozzles when I require only one.

Unfortunately, I cannot provide the project file but I can drop you any settings and answer any questions to help you narrowing it down.

Ghostkeeper, 2019-06-04:
Maybe you could load in two cubes, one for each extruder, and save that as a project? It's more about the settings than the actual model, most likely.

thesec, 2019-06-04:
Sure - find the project (single calibration cube on the left extruder as my repro case, did not try other combinations yet) attached:
[test.3mf.zip](https://github.com/Ultimaker/Cura/files/3252132/test.3mf.zip)

The generated g-code starts with 
```
;FLAVOR:Marlin
;TIME:894
;Filament used: 0.32642m, 0m
;Layer height: 0.215
;MINX:87.875
;MINY:136.375
;MINZ:0.25
;MAXX:122.125
;MAXY:160.625
;MAXZ:14.655
;Generated with Cura_SteamEngine 4.1.0-BETA
T0
M104 S256
M105
M109 S256
M82 ;absolute extrusion mode
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z5 F200 ;safety Z axis movement
M190 S85 ;wait for the bed to heat
G92 E0
T0 ;switch to the left extruder
M109 T0 S256 ;wait for the hotend to heat
G92 E0 ;zero the extruded length
G1 F72 E15 ;extrude 15mm of feed stock
G92 E0
G4 P2000 ;stabilize hotend's pressure
G92 E0
G1 F2400 E-5
;LAYER_COUNT:69
;LAYER:0
...
```

Ghostkeeper, 2019-06-04:
Your project file contains a printer definition that you manually installed into Cura. Where can I find this mod? Without it, the project file cannot be loaded.

thesec, 2019-06-04:
Ah, mea culpa - is there an option to export/save it as a single file ... there was something .ini'ish being mentioned? Which settings' folders and files will it be, if not?

Ghostkeeper, 2019-06-06:
There is no export function for definition files. They are meant to be distributed only via the installation, which is why you had to copy-paste something into your installation folder to make it work.

Basically I need whatever you copy-pasted into your installation folder. I think that includes a file like `bcn3dsigma.def.json` and maybe something like `bcn3dsigma_extruder_left.def.json` and `bcn3dsigma_extruder_right.def.json` and perhaps a mesh file for the build plate (but I can hack that out).

thesec, 2019-06-08:
I am currently trying to bootstrap the Sigma's files from scratch with their "Cura" getting buggier ever since they branched it off ... 🤐 ... copy / paste happening from your Cura's default profiles as I find it matching ... 

I attached 
[5793.zip](https://github.com/Ultimaker/Cura/files/3268585/5793.zip) the definitions and current configs including the bed mesh for you to identify the bug's cause.

digitalfrost, 2022-07-27:
**Stale ticket.**
Can this be closed, since this refers to Cura 4.1b and we are now on 5.0 and the label is **Won't Fix/Do** ?

closed 2023-09-29 by github-actions[bot]